LibreDWG/libredwg #235: How to get line type, such as Dotted line, Dashed line tiny

zhengf312, 2020-06-21:
I found libreDWG can get "BYLAYER"/"BYBLOCK"/"CONTINUOUS" line type from dwg file. But i can't find others line type, such as  "Dotted line"/"Dashed line tiny". How to get it? 

if (!strcasecmp (pair->value.s, "BYLAYER"))
                                  flags = 0;
                                if (!strcasecmp (pair->value.s, "BYBLOCK"))
                                  flags = 1;
                                if (!strcasecmp (pair->value.s, "CONTINUOUS"))
                                  flags = 2;
                                dwg_dynapi_common_set_value (
                                    _obj, "ltype_flags", &flags, 0);

rurban, 2020-06-21:
The ltype property is a handle to the a LTYPE entry. The description you posted is from the DXF reader, which handles it different.

See https://github.com/LibreDWG/libredwg/blob/master/src/common_entity_handle_data.spec

If ENTITY.ltype_flags < 3 then it's one of the 3 special cases, if 3 follow the handle.

zhengf312, 2020-06-22:
> The ltype property is a handle to the a LTYPE entry. The description you posted is from the DXF reader, which handles it different.
>
> See [...]

Thanks!

closed 2020-06-22 by rurban