LibreDWG/libredwg #236: About SECTION_APPINFOHISTORY error.

zhengf312, 2020-06-22:
I have a R2007 dwg file, I try to open it with LibreDWG API.
but 'dwg_read_file' fuction return 258:
 'no THUMBNAIL Image Data' error return 2, and 'read_2007_section_appinfohistory' fuction  return 256. LOG_INFO ("%s section not found\n", "AppInfoHistory").

I am not familiar with dwg file data structure. Does every R2007 dwg file have SECTION_APPINFOHISTORY keyword(or data structure)?

In addition, I noticed that for R2007 dwg, LibreDWG use decode_r2007.c to decode it separately, why other versions do not need to decode separately?

rurban, 2020-06-22:
INFO is just an information, not an error.
The AppInfoHistory section is optional, and we cannot parse it yet. We just store it asis.

The r2007 structure is different from the r2000 and r2004. 
Why 2 .c files? This has historic reasons. r2007 support was added seperately by an individual contributor. The goal is to merge it back into decode.c, because the differences are not that big. But for now it works fine, and the priority to merge it back is low.

zhengf312, 2020-06-22:
But it return error DWG_ERR_CRITICAL. when AppInfoHistory is not be found.

error = read_data_section (&sec_dat, dat, sections_map, pages_map,
                             SECTION_APPINFOHISTORY);
  if (error >= DWG_ERR_CRITICAL || !sec_dat.chain)
    {
      LOG_INFO ("%s section not found\n", "AppInfoHistory");
      if (sec_dat.chain)
        free (sec_dat.chain);
      return error;
    }

zhengf312, 2020-06-22:
> INFO is just an information, not an error.
> The AppInfoHistory section is optional, and we cannot parse it yet. We just store it asis.
> THUMBNAL data is also optional.
> These error codes are not critical, just informative.
> 
> The r2007 structure is different from the r2000 and r2004.
> Why 2 .c files? This has historic reasons. r2007 support was added seperately by an individual contributor. The goal is to merge it back into decode.c, because the differences are not that big. But for now it works fine, and the priority to merge it back is low.

But it return error DWG_ERR_CRITICAL. when AppInfoHistory is not be found.

error = read_data_section (&sec_dat, dat, sections_map, pages_map,
SECTION_APPINFOHISTORY);
if (error >= DWG_ERR_CRITICAL || !sec_dat.chain)
{
LOG_INFO ("%s section not found\n", "AppInfoHistory");
if (sec_dat.chain)
free (sec_dat.chain);
return error;
}

rurban, 2020-06-22:
Oh, that's a bug then. This section is optional I think, but I'll check.
Thanks!

rurban, 2020-06-22:
Fixes in master

closed 2020-06-22 by rurban