LibreDWG/libredwg #237: How to get MText's rotation value

zhengf312, 2020-06-24:
I found Dwg_Entity_TEXT have rotation value(/*!< DXF 50 */), but can't find the rotation value in Dwg_Entity_MTEXT struct.

In Dwg_Entity_MTEXT struct, as follows,
BITCODE_BD *column_heights;/*!< DXF 50 */

The column_heights key use 'DXF 50', after  testing, I find column_heights  is not rotation.

so I want to know how to get rotation value in In Dwg_Entity_MTEXT struct.

rurban, 2020-06-24:
The MTEXT.x_axis_dir  11 vector defines the rotation. Normalize and take the cosinus. I'm not sure if it's always normalized to length 1.0, looks like so.

zhengf312, 2020-06-24:
> The MTEXT.x_axis_dir 11 vector defines the rotation. Normalize and take the cosinus. I'm not sure if it's always normalized to length 1.0, looks like so.

rurban, thanks! . I have been taked a long time for seaching it, thank you for helping me many times

closed 2020-06-24 by zhengf312