LibreDWG/libredwg #238: Another question, about font_file and bigfont_file in Dwg_Object_STYLE struct.

zhengf312, 2020-06-24:
I want to get my MTEXT's font name. my codes are as follows,

Dwg_Entity_MTEXT* text = obj->tio.entity->tio.MTEXT;
 Dwg_Object_STYLE *style = text->style->obj->tio.object->tio.STYLE;

 string stylename = style->font_file;
 if(!RS_COMMON::is_str_utf8(style->font_file))
 {
      std::wstring_convert<std::codecvt_utf8<char16_t>,char16_t> ucs2conv;
      stylename = ucs2conv.to_bytes((char16_t*)text->style->obj->name);
}

But I find the 'stylename ' is always one letter,such as 't' or other letter. I think it should be full font name, such as 'romans'.

Maybe there is some problems with my codes or with my understanding..

I need some helps, sincere thanks.

rurban, 2020-06-24:
You have to check the dwg version first. Since r2007 most texts are ucs2, before ASCIIZ in the respective codepage.

zhengf312, 2020-06-24:
> most texts are ucs2

Ok, I try again.

zhengf312, 2020-06-24:
> You have to check the dwg version first. Since r2007 most texts are ucs2, before ASCIIZ in the respective codepage.
> 
> the dwg texts are never utf8, but the new DXF texts are.

if(data.dwg->header.from_version >= R_2007)
        {
            std::wstring_convert<std::codecvt_utf8<char16_t>,char16_t> ucs2conv;
            stylename = ucs2conv.to_bytes((char16_t*)style->font_file);
        }

Yes, I have get right name, Thanks.

closed 2020-06-24 by zhengf312