LibreDWG/libredwg #250: autogen.sh output error

zhangxinguo1491625, 2020-07-19:
I'm trying to run the master version libredwg on ubuntu18.04, but when I run the autogen.sh , it outputs some error:
++ rm -rf autom4te.cache config.status
++ test -d .git -a -f build-aux/git-version-gen
++ git fetch --tags
++ rm .tarball-version
rm: cannot remove '.tarball-version': No such file or directory
+++ build-aux/git-version-gen .tarball-version
++ v=
++ echo
++ echo
++ set -e
++ autoreconf --install --symlink -I m4
build-aux/git-version-gen: WARNING: .tarball-version is missing or damaged
configure.ac:22: error: AC_INIT should be called with package and version arguments
/usr/share/aclocal-1.15/init.m4:29: AM_INIT_AUTOMAKE is expanded from...
configure.ac:22: the top level
autom4te: /usr/bin/m4 failed with exit status: 1
aclocal: error: echo failed with exit status: 1
autoreconf: aclocal failed with exit status: 1

rurban, 2020-07-19:
See #247. git is the culprit.
Run it again.

rurban, 2020-07-19:
Despite this being a valid but seldom error (none if the smokers ever ran into this), I cannot think of a way to fix this external error. If git fails it is better to fails than to silently ignore it or fix it up and have no .version

closed 2020-07-19 by rurban