LibreDWG/libredwg #257: A double free in free.c:1314

seviezhou, 2020-08-01:
## System info

Ubuntu X64, gcc (Ubuntu 5.5.0-12ubuntu1), dwg2dxf (latest master [4b99ed](https://github.com/LibreDWG/libredwg/commit/4b99edb0ea26e99ef65c5fe68670e6b1f9382d44))

## Configure

CFLAGS="-g -fsanitize=address" LDFLAGS="-fsanitize=address" ./configure

## Command line

./programs/dwg2dxf -b -m ./double-free-dwg_free-free-1314 -o /dev/null

## AddressSanitizer output

```
=================================================================
==39755==ERROR: AddressSanitizer: attempting double-free on 0x60400000dfd0 in thread T0:
    #0 0x7f2009c8a2da in free (/usr/lib/x86_64-linux-gnu/libasan.so.2+0x982da)
    #1 0x555ce2572f6f in dwg_free /home/seviezhou/libredwg/src/free.c:1314
    #2 0x555ce22818c2 in main /home/seviezhou/libredwg/programs/dwg2dxf.c:363
    #3 0x7f2009484b96 in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x21b96)
    #4 0x555ce22828f9 in _start (/home/seviezhou/libredwg/programs/dwg2dxf+0xa528f9)

0x60400000dfd0 is located 0 bytes inside of 40-byte region [0x60400000dfd0,0x60400000dff8)
freed by thread T0 here:
    #0 0x7f2009c8a2da in free (/usr/lib/x86_64-linux-gnu/libasan.so.2+0x982da)
    #1 0x555ce2570682 in dwg_free_header_vars /home/seviezhou/libredwg/src/free.c:1169
    #2 0x555ce2570682 in dwg_free /home/seviezhou/libredwg/src/free.c:1268
    #3 0x10100020201007e  (<unknown module>)

previously allocated by thread T0 here:
    #0 0x7f2009c8a7aa in __interceptor_calloc (/usr/lib/x86_64-linux-gnu/libasan.so.2+0x987aa)
    #1 0x555ce22ecd7d in dwg_decode_handleref /home/seviezhou/libredwg/src/decode.c:4568
    #2 0x303ffffffffff  (<unknown module>)

SUMMARY: AddressSanitizer: double-free ??:0 free
==39755==ABORTING
```

## POC

[double-free-dwg_free-free-1314.zip](https://github.com/LibreDWG/libredwg/files/5010784/double-free-dwg_free-free-1314.zip)

rurban, 2020-08-01:
Cannot repro yet

seviezhou, 2020-08-02:
Strange, I can still reproduce it with the latest master [39ef943](https://github.com/LibreDWG/libredwg/commit/39ef9439f35111170258cf57cc0c58eabc4748c7). Do you use the sanitizer, or may be it is because I am using the GCC 5?  If it really cannot be reproduce, you can close this issue.

Re-upload POC: 
[double-free-dwg_free-free-1314.zip](https://github.com/LibreDWG/libredwg/files/5011516/double-free-dwg_free-free-1314.zip)

```
Reading DWG file ./double-free-dwg_free-free-1314
Warning: checksum: 0x28751255 (calculated) mismatch

ERROR: Invalid EED size 59410 > 5
ERROR: bit_read_RC buffer overflow at 11
ERROR: bit_read_RC buffer overflow at 11
Warning: check_CRC mismatch 4410-4424 = 14: D8BC <=> C37B

Warning: Unstable Class object 504 TABLESTYLE (0xfff) 42/0
Warning: TODO TABLESTYLE r2010+ missing fields
Warning: Unstable Class object 505 MATERIAL (0x481) 45/0
Warning: Unstable Class object 505 MATERIAL (0x481) 46/0
Warning: Unstable Class object 505 MATERIAL (0x481) 47/0
ERROR: Invalid object handle 0.0.0 at pos @2.2
ERROR: bit_read_RC buffer overflow at 54
ERROR: bit_read_RC buffer overflow at 54
ERROR: bit_read_RC buffer overflow at 54
ERROR: bit_read_RC buffer overflow at 54
ERROR: bit_read_RC buffer overflow at 54
Warning: Wrong POLYLINE.layer 0
Warning: POLYLINE.layer is vertex[0] 10, shift em, NULL seqend
Warning: Empty POLYLINE.seqend
Warning: Empty POLYLINE.vertex[4]
Warning: check_CRC mismatch 4351-4408 = 57: 8A9D <=> 9B4D

ERROR: Preview overflow > 90
ERROR: Some section size or address out of bounds
Warning: Skip empty section 0 AcDb:Template
ERROR: Template section not found

Warning: AcDs _data_ segment yet unhandled
Warning: AcDs _data_ segment yet unhandled
Warning: AcDs schdat segment yet unhandled
Warning: Object handle not found, 33/21 in 143 objects
Writing DXF file /dev/null
ERROR: File not overwritten: /dev/null, use -y.

WRITE ERROR /dev/null
WRITE ERROR /dev/null
Warning: TODO TABLESTYLE r2010+ missing fields
Warning: AcDs schdat segment yet unhandled
=================================================================
==5142==ERROR: AddressSanitizer: attempting double-free on 0x60400000dfd0 in thread T0:
    #0 0x7f15d03102da in free (/usr/lib/x86_64-linux-gnu/libasan.so.2+0x982da)
    #1 0x7f15cdec5544 in dwg_free /home/seviezhou/libredwg/src/free.c:1314
    #2 0x55927a093124 in main /home/seviezhou/libredwg/programs/dwg2dxf.c:363
    #3 0x7f15cc89ab96 in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x21b96)
    #4 0x55927a0917c9 in _start (/home/seviezhou/libredwg/programs/.libs/dwg2dxf+0x27c9)

0x60400000dfd0 is located 0 bytes inside of 40-byte region [0x60400000dfd0,0x60400000dff8)
freed by thread T0 here:
    #0 0x7f15d03102da in free (/usr/lib/x86_64-linux-gnu/libasan.so.2+0x982da)
    #1 0x7f15cdebe599 in dwg_free_header_vars /home/seviezhou/libredwg/src/free.c:1169
    #2 0x7f15cdec4712 in dwg_free /home/seviezhou/libredwg/src/free.c:1268
    #3 0x55927a093124 in main /home/seviezhou/libredwg/programs/dwg2dxf.c:363
    #4 0x7f15cc89ab96 in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x21b96)

previously allocated by thread T0 here:
    #0 0x7f15d03107aa in __interceptor_calloc (/usr/lib/x86_64-linux-gnu/libasan.so.2+0x987aa)
    #1 0x7f15cd3cac93 in dwg_decode_handleref /home/seviezhou/libredwg/src/decode.c:4568
    #2 0x7f15cd3f0233 in dwg_decode_header_variables /home/seviezhou/libredwg/src/header_variables.spec:235
    #3 0x7f15cd357164 in read_2004_section_header /home/seviezhou/libredwg/src/decode.c:2740
    #4 0x7f15cd3b10fb in decode_R2004 /home/seviezhou/libredwg/src/decode.c:3676
    #5 0x7f15cd2b7f0c in dwg_decode /home/seviezhou/libredwg/src/decode.c:242
    #6 0x7f15cd2932a0 in dwg_read_file /home/seviezhou/libredwg/src/dwg.c:251
    #7 0x55927a092ba8 in main /home/seviezhou/libredwg/programs/dwg2dxf.c:258
    #8 0x7f15cc89ab96 in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x21b96)

SUMMARY: AddressSanitizer: double-free ??:0 free
==5142==ABORTING
```

rurban, 2020-08-02:
Oh, now I can repro it. 
It's -m/--minimal so it's the HANDSEED only.

closed 2020-08-02 by rurban