LibreDWG/libredwg #266: LibreDWG for Unreal Engine 4.

tincivilfx, 2020-08-05:
Hi,

We would like to know if we can use LibreDWG to import meshes from .dwg to the real time engine for rendering.

Thanks for your work on this so far!

rurban, 2020-08-05:
No idea how the Unreal Engine represents 3D data. But you certainly can use LibreDWG to import DWG or DXF and then convert some of the 3D objects to your format.

eryar, 2020-08-05:
If your 3d in DWG/DXF in the form of 3DFACE or MESH, I think you can use LibreDWG to convert these 3d data to Unreal Engine.
If the 3d in DWG/DXF is 3DSOLID, I think you need a geometry kenerl to convert ACIS of 3DSOLID.

tincivilfx, 2020-08-05:
Thanks both for detailed comments. My .dwg files would always contain valid 3d meshes.
 
Is there an existing framework in the library to pull out: vertices positions, triangles, normals, uv0, vertexcolors, tangents of a given mesh (layer)?

rurban, 2020-08-06:
Nope, no API around objects so far. Only raw access to the fields and vectors. dwg_api.h has s little bit of vertex accesses, but not much. And geom.c does little bit of OCS to UCS conversion for SVG.

Begalov, 2020-09-01:
I can recommend blender as an intermediary for DXF import, model processing and export to OBJ or FBX