LibreDWG/libredwg #267: dwg2SVG is creating empty file

maciekglowacki, 2020-08-19:
Hi, I tried using dwg2SVG commandline utility, however it failed with all of my dwg files resulting in generating empty svg files. 

Is that a known problem?

rurban, 2020-08-19:
No, all the tests pass and are even tested for correctness with a SVG linter. Please describe the problem better.

maciekglowacki, 2020-08-20:
I attach logs from -v1 used to convert these files. I want to point out that I converted the same files to svgs using commercial program and the conversion was successful. 

[image]
[image]

In this case conversion resulted in empty svg file.

rurban, 2020-08-20:
No screenshots, attach the dwg files zipped please.
This seems to be an unsupported DWG file, with broken sections. Maybe related to #144

maciekglowacki, 2020-08-21:
Zipped files: [dwg.zip](https://github.com/LibreDWG/libredwg/files/5107334/dwg.zip)

Can I repair these files on my own?

maciekglowacki, 2020-08-26:
Thank you very much for your help. I will stay with this non-free converter for now :D

rurban, 2021-04-08:
1. Duplicate of GH #144
2. the other consists only of proprietary entities

closed 2021-04-08 by rurban