LibreDWG/libredwg #268: Dockfile building

Begalov, 2020-09-01:
Tried to build Dockefile of cloned repo
Breaks on next lines at STEP 1 (Step 5)
rax     0xc000638000
rbx     0xc0003370d0
rcx     0xa
rdi     0x8672790
rsi     0x78cfb50
rbp     0xc0006bda48
rsp     0x78cf8e0
r8      0xa
r9      0xc000637ff0
r10     0x8630fb0
r11     0x8630010
r12     0x5350000
r13     0x1
r14     0x1
r15     0x1
rip     0x7e13a4f9
rflags  0x10293
cs      0x33
fs      0x53
gs      0x2b

Full console output https://paste.debian.net/1162131/

rurban, 2020-09-02:
First: This Dockerfile misses the pcre2 dependency.

Second: The HELIX unit-tests fails, which should be ignored, because it is not stable.
But in the docker case the SEGV is fatal, whilst with testing in the shell it is not. I have no idea on special such docker signal behaviors.

rurban, 2020-09-02:
podman works fine, just docker is broken.

rurban, 2020-09-03:
Even without docker it crashes now in dxf_test at 2018/Helix.dwg. See eg https://travis-ci.org/github/LibreDWG/libredwg/jobs/723376835

closed 2020-11-05 by rurban