LibreDWG/libredwg #269: add gambas bindings

rurban, 2020-09-02:
see the `work/gambas` branch

rurban, 2020-12-07:
gambas support is done in libredwg. gambas provides no pkg-config, so we cannot get its headers. They are not in `/usr/include/gambas3` nor `/usr/share/gambas3/include`. So we cannot include it here, it must be under the gambas srcdir.
See  https://github.com/LibreDWG/gambas3-bindings

closed 2020-12-07 by rurban

belmotek, 2020-12-07:
Ok, I will see this whit the gambas development team in order to use it.

We (GauchoCAD) actually are using terminal tool dwgread > .json then 
decode json to made the objects collections then we write a output 
.json  file then convert .json to .dwg

Thank you very much for preparing the Gambas bindings.

Martín Belmonte

GauchoCAD team