LibreDWG/libredwg #271: Use dwgread.exe to save as Geojson file, the Text is not converted to utf-8

chensccode, 2020-10-13:


rurban, 2020-10-13:
from which DWG version? before r2007, or after?

chensccode, 2020-10-13:
> from which DWG version? before r2007, or after?

from R2007  after.
Here is the test result 
r2010->  "Text": "KmՋo",
r2004 -> "Text": "测试ok",

chensccode, 2020-10-20:
> from which DWG version? before r2007, or after?

in function "dwg_geojson_feature" in "out_geojson.c" source file, I add some to change che Text string to UTF-8, and result is OK , about 468 line:

```C
  else if (obj->type == DWG_TYPE_TEXT)
    {
      Dwg_Entity_TEXT *_obj = obj->tio.entity->tio.TEXT;
      // PAIR_S (Text, _obj->text_value)
      if (IS_FROM_TU (dat))
        {
          char *utf8 = bit_convert_TU ((BITCODE_TU)_obj->text_value);
          PAIR_S (Text, utf8)
          free (utf8);
        } 
      else
        {
          PAIR_S (Text, _obj->text_value)
        }
    }
  else if (obj->type == DWG_TYPE_MTEXT)
    {
      Dwg_Entity_MTEXT *_obj = obj->tio.entity->tio.MTEXT;
      // PAIR_S (Text, _obj->text);
      if (IS_FROM_TU (dat))
        {
          char *utf8 = bit_convert_TU ((BITCODE_TU)_obj->text);
          PAIR_S (Text, utf8)
          free (utf8);
        } 
      else
        {
          PAIR_S (Text, _obj->text)
        }
    }
```

closed 2020-10-30 by rurban