LibreDWG/libredwg #273: Dockerfile fails to build correctly

matthewbroadwaynavenio, 2020-10-21:
I cloned the repository and ran `docker build .` The image did not build successfully and exited with:
```
...
...
make  all-am
make[4]: Entering directory '/app/libredwg-0.11/bindings/python'
  CC       _LibreDWG_la-swig_python.lo
  CCLD     _LibreDWG.la
ar: `u' modifier ignored since `D' is the default (see `U')
make[4]: Leaving directory '/app/libredwg-0.11/bindings/python'
make[3]: Leaving directory '/app/libredwg-0.11/bindings/python'
make[3]: Entering directory '/app/libredwg-0.11/bindings'
/usr/bin/swig -perl5 -I/app/libredwg-0.11/include -o perl/LibreDWG.c ./dwg.i
/app/libredwg-0.11/include/dwg.h:5083: Warning 314: 'ref' is a perl keyword
/app/libredwg-0.11/include/dwg.h:8470: Warning 314: 'ref' is a perl keyword
make[3]: *** No rule to make target 'perl/LibreDWG.xs', needed by 'all-am'.  Stop.
make[3]: Leaving directory '/app/libredwg-0.11/bindings'
make[2]: Leaving directory '/app/libredwg-0.11/bindings'
make[2]: *** [Makefile:459: all-recursive] Error 1
make[1]: Leaving directory '/app/libredwg-0.11'
make[1]: *** [Makefile:669: all-recursive] Error 1
make: *** [Makefile:552: all] Error 2
The command '/bin/sh -c tarxz=`curl --silent 'https://ftp.gnu.org/gnu/libredwg/?C=M;O=D' | grep '.tar.xz<' |          head -n1|sed -E 's/.*href="([^"]+)".*/\1/'`;     echo "latest release $tarxz";     curl --silent --output "$tarxz" https://ftp.gnu.org/gnu/libredwg/$tarxz &&     tar xfJ "$tarxz" &&     cd `echo "$tarxz" | sed s'/.tar.xz//'` &&     ./configure --enable-release &&     make &&     mkdir install &&     make install DESTDIR="$PWD/install" &&     make check DOCKER=1 DESTDIR="$PWD/install"' returned a non-zero code: 2
```

rurban, 2020-10-21:
Do you need the Perl or Python bindings?
If not just use --disable-bindings on the configure line

matthewbroadwaynavenio, 2020-10-21:
thanks for the quick reply.
using `--disable-bindings` did let the build get further. It then crashed in the COPY stage since the library is at `/app/libredwg-0.11` not at `/app/libredwg` where the COPY command is expecting.

I've made the following modifications to the Dockerfile to get it to build. I could open a pull request if you like?
```patch
diff --git a/Dockerfile b/Dockerfile
index 649a5036..6f5015e1 100644
--- a/Dockerfile
+++ b/Dockerfile
@@ -1,4 +1,4 @@
-# podman/docker build -t libredwg . 
+# podman/docker build -t libredwg .
 ############################
 # STEP 1 build package from latest tar.xz
 ############################
@@ -18,16 +18,18 @@ RUN apt-get update && \
     make && \
     make install && \
     cd /libxmlInstall && \
-    rm -rf gg libxml2-$LIBXML2VER.tar.gz libxml2-$LIBXML2VER            
+    rm -rf gg libxml2-$LIBXML2VER.tar.gz libxml2-$LIBXML2VER
 WORKDIR /app
 RUN tarxz=`curl --silent 'https://ftp.gnu.org/gnu/libredwg/?C=M;O=D' | grep '.tar.xz<' | \
          head -n1|sed -E 's/.*href="([^"]+)".*/\1/'`; \
     echo "latest release $tarxz"; \
     curl --silent --output "$tarxz" https://ftp.gnu.org/gnu/libredwg/$tarxz && \
-    tar xfJ "$tarxz" && \
-    cd `echo "$tarxz" | sed s'/.tar.xz//'` && \
-    ./configure --enable-release && \
-    make && \
+    mkdir libredwg && \
+    tar -C libredwg --xz --strip-components 1 -xf "$tarxz" && \
+    rm "$tarxz" && \
+    cd libredwg && \
+    ./configure --disable-bindings --enable-release && \
+    make -j `nproc` && \
     mkdir install && \
     make install DESTDIR="$PWD/install" && \
     make check DOCKER=1 DESTDIR="$PWD/install"

```

rurban, 2020-10-21:
ah yes, please.
I never tested it properly

2020, 12:56:

closed 2020-10-22 by matthewbroadwaynavenio