LibreDWG/libredwg #275: Creation of invalid dxf files

matthewbroadwaynavenio, 2020-10-22:
I have collected together many dwg files freely available online and compared the quality of the LibreDWG dwg->dxf conversion with the ODAFileConverter conversion. I'm using the ezdxf python library to read in the results.

Many of the dxf files produced by libreDWG cannot be opened in AutoCAD because they are invalid in some way but it is difficult to determine the cause using AutoCAD. ezdxf gives more detailed error information which may be useful for fixing these issues.

many files have an issue where an x coordinate is provided with no corresponding y coordinate. Examples include 
[missing_y.zip](https://github.com/LibreDWG/libredwg/files/5422402/missing_y.zip)
obtained from [here](https://knowledge.autodesk.com/support/autocad/downloads/caas/downloads/content/autocad-sample-files.html) as dwg and converted using LibreDWG.
43 of 89 dwg files that I tested have this exact issue. I can share more examples if that would be helpful.
see https://github.com/mozman/ezdxf/issues/256 for details

Another very common issue with dxf files produced with libreDWG is that invalid (zero vector) start_tangents are provided for splines. A workaround has been implemented into ezdxf to discard these invalid values but AutoCAD refuses to load these files
an example of this issue is
[Ceco.NET-Architecture-Tm-272.dxf.zip](https://github.com/LibreDWG/libredwg/files/5422438/Ceco.NET-Architecture-Tm-272.dxf.zip)
obtained from [here](https://ceco.net/autocad-blocks/architecture/top-plan-view/autocad-drawing-villa-savoye-le-corbusier-second-floor-rooftop-dwg-dxf-272) and converted using LibreDWG
see https://github.com/mozman/ezdxf/issues/252 for details

A rarer issue is the one found from the color wheel file from [here](https://knowledge.autodesk.com/support/autocad/downloads/caas/downloads/content/autocad-sample-files.html).
[colorwh.dxf.zip](https://github.com/LibreDWG/libredwg/files/5422481/colorwh.dxf.zip)
(It looks like a newline character should have been escaped but wasn't? see https://github.com/mozman/ezdxf/issues/256#issuecomment-714413342)
again, see https://github.com/mozman/ezdxf/issues/256 for details

mozman, 2020-10-24:
Issues of `lineweights.dxf`:

1. The TEXT entity has some duplicated tags:
[image]

- group code 7 in subclass AcDbEntity is not necessary, it is corrrect in the AcDbText subclass
- some invalid vertex group codes 10 and one group code 31
- the group code 41 defines the ratio of height to width and therefore a value of 0 is not valid, but maybe ignored by AutoCAD

2. The LINE entity has an invalid lineweight value:
[image]

Valid lineweight values, this is not documented by the DXF reference and Autodesk products are very picky:

-3, -2, -1, 0, 5, 9, 13, 15, 18, 20, 25, 30, 35, 40, 50, 53, 60, 70, 80, 90, 100, 106, 120, 140, 158, 200, 211

Mapping from DWG lineweight enum to DXF lineweight (from [libdxfrw](https://github.com/LibreCAD/libdxfrw)):

```Python
DXF_LINE_WEIGHT = {
    0: 0,
    1: 5,
    2: 9,
    3: 13,
    4: 15,
    5: 18,
    6: 20,
    7: 25,
    8: 30,
    9: 35,
    10: 40,
    11: 50,
    12: 53,
    13: 60,
    14: 70,
    15: 80,
    16: 90,
    17: 100,
    18: 106,
    19: 120,
    20: 140,
    21: 158,
    22: 200,
    23: 211,
    29: -1,  # BYLAYER
    30: -2,  # BYBLOCK
    31: -3,  # DEFAULT
}
```

rurban, 2020-10-28:
colorwh.dxf also is missing dxf_fixup_string to convert \r\n to ^M^J in $KEYWORDS.
"...(AutoCAD Color^M^J Index"
The SummaryInfo texts are also completely wrong.

rurban, 2020-10-28:
5cc9a6e50d31125c6fa329cfa2270d853f9b40b8 fixed the SummaryInfo TU16 texts. 2007/colorwh converts now fine to dxf.

rurban, 2020-10-28:
colorwh and lineweights can now be imported by Acad.
working on blocks_and_tables_-_metric

rurban, 2020-10-28:
3 additional DXF problems (blocks_and_tables_-_metric)
* LAYER Defpoints apparently requires an empty plotflags 290 with 0.
* Entities not allowed in *Paper_Space block definition on line 5884.
* The following error was encountered while reading in ATTRIB starting at line 75764:
Unexpected DXF group code: 74 Invalid or incomplete DXF input -- drawing discarde

mozman, 2020-10-29:
- I can confirm: AutoCAD requires group code 290 (plotflag) set to 0 for layer `Defpoints` and the layer `Defpoints` is mandatory
- The content of the special blocks `*Model_Space` and `*Paper_Space` (the active paper space) are stored exclusive in the ENTITIES section. A not required but AutoCAD like setting: entities of the paper space has group code 67 set to 1 (0 for model space), but the owner tag (330) also references the corresponding BLOCK_RECORD entry. ENTITIES section: AutoCAD store all model space entities at first, followed by the paper space entities.
- ATTRIB: group code 74 (vertical align) should be located in the subclass `AcDbAttribute` (ATTDEF: `AcDbAttributeDefinition`).

rurban, 2020-10-30:
current state:

work_r12

Unknown header variable $DIMSAV ignored on line 200.
Error 372 in drawing header on line 340.
Invalid or incomplete DXF input -- drawing discarded.

Leader_r14

Error 372 in drawing header on line 428.
Invalid or incomplete DXF input -- drawing discarded.

block...metric 2007

Entities not allowed in *Paper_Space block definition on line 5884. (GH #277)
The following error was encountered while reading
in BLOCKSCALEACTION starting at line 121170:
Missing DXF group code: 1
Invalid or incomplete DXF input -- drawing discarded. (GH #278)

tablet_2007

Entities not allowed in *Paper_Space block definition on line 13364. (GH #277)
The following error was encountered while reading
in HATCH starting at line 117168:
Error: expected group code 97
Invalid or incomplete DXF input -- drawing discarded. (GH #280)

sun_and_sky_demo_2010

The following error was encountered while reading
in 3DSOLID starting at line 3742:
Premature end of object
Invalid or incomplete DXF input -- drawing discarded.

5151-024_2010

The following error was encountered while reading
in HATCH starting at line 22344:
Error: expected group code 97
Invalid or incomplete DXF input -- drawing discarded.  (GH #280)

Dynblocks_2018

The following error was encountered while reading
in AcDbRotatedDimension starting at line 1189414:
Invalid dimension block name.
Invalid or incomplete DXF input -- drawing discarded.

rurban, 2020-11-03:
As of 069f35676c65c34158dbd034f6313c61c628a24d we have these dxfin errors.
Wih the MULTILEADER fixes acad still crashes silently

Fatal:
* 3DSOLID: Premature end of object
* MULTILEADER: Premature end of object
* VERTEX: Bad visibility value 2 (r2000, r2004)
* Error: Block "*Paper_Space" was not defined
* MLEADERSTYLE: DXF read error
* LAYER: Invalid AcDbLayerTableRecord plot flag for DEFPOINTS layer (Publish_2004)
* PDFUNDERLAY: Unexpected DXF group code: 170

Warning:
* Entities not allowed in *Paper_Space block definition on line 2224.

naufraghi, 2020-11-03:
Thanks for the great work, I've found some issues on:

- the `41` and `42` group codes of the `INSERT` entity
- the `10` group code of the `TEXT` or `MTEXT` entity

I'll have a look at the issue and confirm if they are fixed once resolved.

rurban, 2020-11-03:
My estimate is about 2 weeks for these. Looked dramatic at first, but now I see the light at the end of the tunnel.

rurban, 2020-11-04:
@mozman Actually AcDbText 41 is the xScale or the width_factor, which defaults to 1.0.
Not some height/width ratio.

rurban, 2020-11-10:
Mostly fixed now, just some very advanced features cannot be imported still. TODO MLEADERSTYLE

rurban, 2020-11-16:
Fixed most issues, only about 10% of DXF imports do fail now. See above for the annoying silent crashes, which I hope to fix later.
Save all your work before DXFIN.

naufraghi, 2020-11-18:
> Thanks for the great work, I've found some issues on:
> 
> * the `10`, `41` and `42` group codes of the `INSERT` entity
> * the `10` group code of the `TEXT` or `MTEXT` entity
> 
> I'll have a look at the issue and confirm if they are fixed once resolved.

I still have problems with the `41` and `42` group codes of the `INSERT` entity [Help: INSERT (DXF)](https://help.autodesk.com/view/OARX/2018/ENU/?guid=GUID-28FA4CFB-9D5E-4880-9F11-36C97578252F)

The codes `41`, `42` and `43` seems to be repeated twice in the output:

```
...
INSERT
  5
3508
330
34E2
100
AcDbEntity
  8
MOZ_PREP_REPORT
284
     0
100
AcDbBlockReference
  2
PITBT20170628-16295785
 10
64.75
 20
70.97857142857129
 30
0.0
 41
0.32857142857143
 42
0.32857142857143
 43
0.32857142857143
 41
0.32857142857143
 42
0.32857142857143
 43
0.32857142857143
  0
DIMENSION
  5
...
```

closed 2020-11-18 by rurban