LibreDWG/libredwg #281: Add xdata into geojson

chensccode, 2020-10-30:
In some scenarios, we will store some data in xdata. The current geojson file does not contain these data. This function can be added in future versions.

```C
#include <stdio.h>
#include <stdlib.h>

#include "dwg.h"
#include "bits.h"

/**
 * get xdata
 * @param obj Dwg_Object
 */
void object_eed(Dwg_Object *obj) {
    Dwg_Object_Object *obj_obj = obj->tio.object;
    for (unsigned i = 0; i < obj_obj->num_eed; i++) {
        Dwg_Eed eed = obj_obj->eed[i];
        if (eed.size) {
            Dwg_Object *appid = dwg_resolve_handle(obj_obj->dwg, eed.handle.value);
            if (appid && appid->fixedtype == DWG_TYPE_APPID) {
                char *name = appid->tio.object->tio.APPID->name;
                char *utf8 = bit_convert_TU((BITCODE_TU) name);
                printf("key: %s \n", utf8);
            }
        }

        if (eed.data) {
            const Dwg_Eed_Data *data = eed.data;
            char *str_utf8 = bit_convert_TU((BITCODE_TU) data->u.eed_0.string);
            printf("%d: %s \n", data->code, str_utf8);
            free(str_utf8);
        }
    }
}

/**
 * read xdata
 * @param filename
 */
void read_xdata(const char *filename) {
    Dwg_Data dwg;
    unsigned int opts = 1;

    dwg.opts = opts;
    dwg_read_file(filename, &dwg);

    for (unsigned i = 0; i < dwg.num_objects; i++) {
        Dwg_Object obj = dwg.object[i];
        Dwg_Object_Entity *entity = obj.tio.entity;

        if (obj.supertype != DWG_SUPERTYPE_ENTITY) {
            continue;
        }
        Dwg_Object *layer = entity->layer ? entity->layer->obj : NULL;
        if (!layer || layer->type != DWG_TYPE_LAYER) {
            continue;
        }
        object_eed(&obj);
    }
    dwg_free(&dwg);
}
```

[xdata.dwg.zip](https://github.com/LibreDWG/libredwg/files/5462649/xdata.dwg.zip)

rurban, 2020-10-30:
Only strings? Without checking the type (the group code).
There need to be a saner logic and API what APPID's to include, and which data to export from it.

Could you add something like this to the wiki?

chensccode, 2020-10-30:
> Only strings? Without checking the type (the group code).
> There need to be a saner logic and API what APPID's to include, and which data to export from it.
> 
> Could you add something like this to the wiki?

The sample codes is used for my one projected, it is simple. I can write a complete code. But I usually don’t use C language, you need to check my code.

rurban, 2020-10-30:
No code, just documentation what APPID's and types will be written to geojson.