LibreDWG/libredwg #296: dxf2json (similar way as dwg2json)

belmotek, 2020-11-16:
Hi, Reini.
We use the tool dwg2json, is great to import dwg files in GauchoCAD. We miss some dxf2json in order to import DXF also.
¿Is planed dxf2json?
Regards.

rurban, 2020-11-16:
dxf2json would be `dxf2dwg $file.dwg && dwgread -o $file.json $file.dwg`
Internally it would be the same. First import the DXF into a DWG, and then write a JSON.

belmotek, 2020-11-16:
Ok, yes actually i do that. but now is only dxf v2000 right?

rurban, 2020-11-17:
True. DXF can be written in every supported version, just DWG not. By going over the intermediate DWG we can only do v2000. This could be a good reason to create such a simple converter. 
But I have more important things to do now. Would be very simple to do based on some existing converter. Would take a PR, but will be deprecated (i.e. built in examples, but not packaged) once 2004-2018  can be written.

closed 2020-11-17 by rurban