LibreDWG/libredwg #297: dwgread -nan value in object: MATERIAL that is correct?

belmotek, 2020-11-16:
Hi,
in dwgread appear -nan 

    {
      "object": "MATERIAL",
      "index": 114,
      "type": 506,
      "handle": [0, 617],
      "size": 59,
      "bitsize": 402,
      "ownerhandle": [12, 2, 511, 106],
      "reactors": [
        [4, 1, 106, 106]
      ],
      "xdicobjhandle": [3, 2, 618, 618],
      "_subclass": "AcDbMaterial",
      "name": "ByLayer",
      "description": "",
      "ambient_color.flag": 0,
      "ambient_color.factor": 0.0,
      "diffuse_color.flag": 0,
      "diffuse_color.factor": 0.0,
      "diffusemap.blendfactor": 0.0,
      "diffusemap.projection": 0,
      "diffusemap.tiling": 0,
      "diffusemap.autotransform": 0,
      "diffusemap.transmatrix": [
        0.00000000000000,
        -0.00000000000000,
        0.00000000000000,
        -nan,
        -nan,
        -nan,
        -nan,
        -nan,
        -nan,
        -nan,
        -nan,
        -nan,
        -nan,
        -nan,
        -nan,
        -nan        
      ],
      "diffusemap.source": 0,
      "specular_color.flag": 0,
      "specular_color.factor": -nan,
      "specularmap.blendfactor": 0.0,
      "specularmap.projection": 0,
      "specularmap.tiling": 0,
      "specularmap.autotransform": 0,
      "specularmap.transmatrix": [
        
      ],
      "specularmap.source": 0,
      "specular_gloss_factor": 0.0,
      "reflectionmap.blendfactor": 0.0,
      "reflectionmap.projection": 0,
      "reflectionmap.tiling": 0,
      "reflectionmap.autotransform": 0,
      "reflectionmap.transmatrix": [
        
      ],
      "reflectionmap.source": 0,
      "opacity_percent": 0.0,
      "opacitymap.blendfactor": 0.0,
      "opacitymap.projection": 0,
      "opacitymap.tiling": 0,
      "opacitymap.autotransform": 0,
      "opacitymap.transmatrix": [
        
      ],
      "opacitymap.source": 0,
      "bumpmap.blendfactor": 0.0,
      "bumpmap.projection": 0,
      "bumpmap.tiling": 0,
      "bumpmap.autotransform": 0,
      "bumpmap.transmatrix": [
        
      ],
      "bumpmap.source": 0,
      "refraction_index": 0.0,
      "refractionmap.blendfactor": 0.0,
      "refractionmap.projection": 0,
      "refractionmap.tiling": 0,
      "refractionmap.autotransform": 0,
      "refractionmap.transmatrix": [
        
      ],
      "refractionmap.source": 0
    },

rurban, 2020-11-16:
That's a bug, but I think I weeded out all nan's already in the released tarballs. 
In a dev version from git a nan is written to find such bugs.
MATERIAL is not stable yet, I would skip it.

belmotek, 2020-11-16:
Okay I'll skip it.
Tanks.

rurban, 2020-11-17:
Will be fixed once MATERIAL goes into stable

closed 2020-11-17 by rurban