LibreDWG/libredwg #303: dwgread on AC1032 files results in empty file

nddipiazza, 2021-01-09:
is this expected?

rurban, 2021-01-09:
Nope, r2018 (AC1032) works fine

nddipiazza, 2021-01-10:
Ah. I see I thought it was empty, but in reality the SummaryInfo section was just empty.

Here is the example I'm using: https://forums.autodesk.com/autodesk/attachments/autodesk/706/1023779/1/AutoCAD%202018%20format.dwg

```
"SummaryInfo": {
    "TITLE": "",
    "SUBJECT": "",
    "AUTHOR": "",
    "KEYWORDS": "",
    "COMMENTS": "",
    "LASTSAVEDBY": "",
    "REVISIONNUMBER": "",
    "HYPERLINKBASE": "",
    "TDINDWG": [ 0, 0 ],
    "TDCREATE": [ 0, 0 ],
    "TDUPDATE": [ 0, 0 ],
    "unknown1": 0,
    "unknown2": 0
  },
```

I was under the impression all dwg files would at least have `TDCREATE` in the `SummaryInfo`. 
But everytime I parse a 2018  DWG file it's empty.

nddipiazza, 2021-01-10:
If that appears to be expected behavior, please close this issue. I'll leave it open in case this is an actual issue. 
I tested using `text.dwg` from the test-files and there is no issue with that.

rurban, 2021-01-14:
The SummaryInfo is rarely filled. TDCREATE is in the HEADER.
Thanks for the sample, but I think it's copyrighted.

closed 2021-01-14 by rurban

nddipiazza, 2021-01-15:
Yeah that's what I'm seeing too. On a huge site with 1000's of dwg, only a few have summary.