LibreDWG/libredwg #310: LINE entity with null layer and wrong start or end point coordinates

shanzhugit, 2021-01-28:
Hi,
Glad to try the libredwg lib and thank you so much for your work.
I hava tried some dwg files and it works fine. But when I use this dwg file I attached with only two LINE entites, I got some wrong data.  The layer object of LINE entity is null, and also the start and end point of Line entities in this dwg file is not correct, very very strange coordinates values,like  [ -3.77695079567220847e+165, -2.9177506778545462e+168, 0.0 ], [ 0.0, -2.86035487996884257e+168, 0.0 ] . Code below:

 Dwg_Object *ms = dwg_model_space_object (&dwg);
 obj = get_first_owned_entity (ms); 
 while (obj)
    {
 if(obj->type == DWG_TYPE_LINE)
      {
        Dwg_Object_LAYER * dwgLayerobj =dwg_get_entity_layer(obj->tio.entity);
        Dwg_Entity_LINE *line = obj->tio.entity->tio.LINE;
}
}
The  dwgLayerobj is null ,and  the line entity also get wrong start or end coordinate.
Could you help to check if this issue can be solved?  I can open it correctly in AutoCAD or Microstation, so I think the file is ok. But I have tried other dwg files by the lib, I can get the correct values of LINE entities.
Thank you~~
[d12007.zip](https://github.com/LibreDWG/libredwg/files/5889282/d12007.zip)

rurban, 2021-01-30:
It's caused by some EED reading error, corrupting the offsets. Investigating...

rurban, 2021-01-31:
Thanks, that was a major error. 
EED layers (code 3) need 8 bytes, not 4. That's why we got the wrong offset and size.
Not many dwg's do have EED layers. None so far.

shanzhugit, 2021-01-31:
Great~~Is there a plan of 0.12.1 release time? @rurban

closed 2021-01-31 by rurban