LibreDWG/libredwg #312: font_file is null When the TEXT font name is not  plain ASCII

shanzhugit, 2021-02-05:
Hi~
For Text entity, if the font name is Arial or others that is plain ASCII, we can get the correct font_file. But when Text entity use Chinese font with Chinese font name  like “黑体”  or "宋体"( btw, their .ttf file names are plain English), it returns null font _file. See the example file attached.
          Dwg_Entity_TEXT *text;
          text = dwg.object[i].tio.entity->tio.TEXT;
          Dwg_Object_STYLE *style = text->style->obj->tio.object->tio.STYLE;
          char *stylename;
          stylename = bit_convert_TU((BITCODE_TU)style->font_file);
Could you help check if the issue can be solved? 
Thank you~~
[Text.zip](https://github.com/LibreDWG/libredwg/files/5934216/Text.zip)

rurban, 2021-02-23:
In the case of the 3rd style, with name "FS" and the EED 
```
EED[0] code: 0 [RC], wstring: len=6 [RS] "SimHei" [TU]
EED[1] code: 71 [RC], long: 34353 [RL]
```
the length of the font_file fields is 0.
So it looks like you want to copy the EED code 0 name over to font_file? This is problematic.

I'd rather add a helper function to get the font_file property either from the font_file field, or the EED, which also adds much more TTF flags. "Simhei" would be the family I guess, 
pitch, charset, and italic and bold flags are in the flag 34353, i.e. 0x8631

labman003, 2023-12-15:
style->font_file is   empty string