LibreDWG/libredwg #313: Support for DWG R2018 ?

jossefaz, 2021-02-09:
Hi

I tried to dwg2dxf a 2018 dwg format.
The output dxf does not open correctly (it does not open at all and seems to be "corrupted" somehow)

As I saw in the documentation the library currently support only few of R2010+

Do you think that 2018+ will be available soon ?

rurban, 2021-02-09:
The docs are wrong. R2018 is supported, and r2018 is the latest known DWG format. (up to AutoCAD 2021)

I'll fix this dxf error, thanks for the testcase.

rurban, 2021-02-09:
Cannot confirm. 2018_3.dxf converted with 0.12.1 dwg2dxf opens fine on my AutoCAD 2018. Without any warning.

jossefaz, 2021-02-10:
Okay. Weird.
I'll check once again.

Thanks for your time and patience !

rurban, 2021-02-10:
Maybe you tried with 0.12, not 0.12.1

closed 2021-02-22 by rurban