LibreDWG/libredwg #318: AutoCAD sample files can't be read

ociule, 2021-02-24:
Hello everyone,

Thanks for the work on this very promising library.

I encountered some errors on reading (using dwgread) proprietary files I can't share (I'm working on a minimal reproducing sample that could be shared eventually) but I get similar errors while trying to read the sample DWG files provided by Autocad, that you can find here:

https://knowledge.autodesk.com/support/autocad/downloads/caas/downloads/content/autocad-sample-files.html

I think these would be a great target to be able to read, as they seem to be quite popular and widely used as teaching examples, and would be a good addition to your tests.

I've tried a few, I can confirm that `civil_example-imperial.dwg` doesn't work, and I'm sure there are others.

Cheers, thanks again, and good luck!

rurban, 2021-02-25:
Confirmed. Several of those r2010 (Gator Beta 2) examples have problems reading some sections. mostly invalid string stream offsets with hi_size, which lead to wrong classes.
I already had them in my extended testing pipeline.

However those problems should not lead to INTERNALERROR, as most entities and objects can be read just fine. Just some strings are missing,

rurban, 2021-02-25:
Fixed in master, thanks.
Can you try again? This wrong string offset was for all r2010 beta1-3 versions.

closed 2021-02-26 by rurban