LibreDWG/libredwg #321: NULL dereference in header_variables_dxf.spec:26

zodf0055980, 2021-03-02:
I found a heap buffer overflow in the current master (7f525b9).

# Configure
CFLAGS="-g -fsanitize=address" LDFLAGS="-fsanitize=address" ./configure

# Command
`./dwg2dxf -o /dev/null -m -y ./poc1`

# ASAN report
```
➜  ./dwg2dxf -o /dev/null -m -y ./poc1 

Reading DWG file ./poc1
ERROR: bit_read_fixed buffer overflow at pos 128, size 152
ERROR: read_file_header Invalid file_header->header_size 9223091660467011385 > MAX_SIZE
ERROR: read_file_header Invalid file_header->file_size 65362 > MAX_SIZE
ERROR: read_file_header Invalid file_header->pages_maxid 9218588057970734863 > MAX_COUNT
ERROR: read_file_header Invalid file_header->pages_amount 1095652868096 > MAX_COUNT
ERROR: read_file_header Invalid file_header->num_sections 9177492712645984768 > MAX_COUNT
Writing DXF file /dev/null
ASAN:DEADLYSIGNAL
=================================================================
==10341==ERROR: AddressSanitizer: SEGV on unknown address 0x000000000010 (pc 0x7ffff3c883cf bp 0x7fffffffcac0 sp 0x7fffffffc1c0 T0)
==10341==The signal is caused by a READ memory access.
==10341==Hint: address points to the zero page.
    #0 0x7ffff3c883ce in dxf_header_write /home/yuan/afl-target/libredwg-test/src/header_variables_dxf.spec:26
    #1 0x7ffff3cbfc7e in dwg_write_dxf /home/yuan/afl-target/libredwg-test/src/out_dxf.c:3621
    #2 0x5555555589bc in main /home/yuan/afl-target/libredwg-test/programs/dwg2dxf.c:336
    #3 0x7ffff1e6bbf6 in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x21bf6)
    #4 0x555555556a89 in _start (/home/yuan/afl-target/libredwg-test/programs/.libs/dwg2dxf+0x2a89)

AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV /home/yuan/afl-target/libredwg-test/src/header_variables_dxf.spec:26 in dxf_header_write
==10341==ABORTING

```
# POC
[poc1.zip](https://github.com/LibreDWG/libredwg/files/6066295/poc1.zip)

rurban, 2021-03-03:
Already fixed with 5d2c75fd5cc7d7cc2e19e158237737bdaa2a9697

zodf0055980, 2021-03-03:
But I test in 8072563, it still have this problem.

rurban, 2021-03-06:
I get this
```
ERROR: bit_read_fixed buffer overflow at pos 128, size 152
seqence_crc64: 0000006D4E0044B8
seqence_key:   7FCB000000004554
compr_crc64:   7F0900FEFE004500
compr_len:     8288
len2:          2130706432
ERROR: Decompression error: offset underflow

num_objects: 0
num_object_refs: 0
ERROR: Failed to read 2007 meta data
ERROR: Failed to decode file: ../test/issues/gh321/poc1.dwg 0x400

READ ERROR 0x400
```

which is good

zodf0055980, 2021-03-06:
I not get error without `ASAN`. Maybe this is asan problem (?).

rurban, 2021-03-06:
Dont think so. I'll let it open for a while

zodf0055980, 2021-03-07:
I think this problem is about `data` in `read_file_header` didn't initialize.
In read_file_header, it creates `data[0x3d8]` but uninitialized. 
https://github.com/LibreDWG/libredwg/blob/625da7ad8b7f6de51e9723ec8e7e7d6714307017/src/decode_r2007.c#L1185
And in this poc, in `bit_read_fixed` in `read_file_header` will detect buffer overflow, so data is still uninitialized.
https://github.com/LibreDWG/libredwg/blob/625da7ad8b7f6de51e9723ec8e7e7d6714307017/src/decode_r2007.c#L1198
https://github.com/LibreDWG/libredwg/blob/f086e9d99f0ce1551348df57d6db9b904c2812be/src/bits.c#L1415-L1424
so `pedata` may have a dirty value to cause this problem.

rurban, 2021-03-07:
Yes, a memset is needed. Nice catch

zodf0055980, 2021-03-08:
Thanks, Could I try to submit these problems to get CVE IDs in version `0.12.3`  ?

rurban, 2021-03-08:
Sure you can. But do it now, as they need way too long. Like over 14 days.

I have over 20 such issues in private also

closed 2021-03-10 by rurban