LibreDWG/libredwg #322: Cannot get rid of AutoCAD recovery mode

timoria21, 2021-03-02:
I've made the simplest DWG containing only one circle and I still get the Recovery Mode when opening it in AutoCAD 2007. From what I can see, LibreDWG is unable to write accurate DWG files, in all cases.

I'm attaching the files, just in case I am missing something. I don't believe that these warnings are meaningful in this respect.

[image]

[image]

[files.zip](https://github.com/LibreDWG/libredwg/files/6066834/files.zip)

rurban, 2021-03-03:
Set the RECOVERAUTO variable

rurban, 2021-03-03:
Fix the wrong type in an APPID, leading to the NOD being ignored/.

rurban, 2021-03-05:
yes. this was it. 14527d3eed139a23a6cd165ed591be8337ad18ba should fix those problems.

rurban, 2021-03-05:
That being said, getting rid of the recovery dialog on acad import is not a goal for this library. We try to keep as much information as possible, and don't simply skip half-known information. acad recovery is doing its thing fine. there are so many unstable objects still, so there will always be a recovery prompt if you don't set to automatic. only with r2000 -> r2000 conversion without any unstable object there will be none

closed 2021-03-05 by rurban

timoria21, 2021-03-05:
I understand your point but nobody will trust a DWG file with this error in AutoCAD, how can you tell that the recovered entities are 100% of the DWG content? Will anybody in the industry use a DWG file with this problem?

If you have any idea of where to start investigating this issue, I will try to help with this cause. Would skipping half-known information help? Where do I start?

Thanks.

rurban, 2021-03-05:
Sorry, but a 100% perfect DWG is a none-goal. Data-loss is a worse problem then a message that the dwg needs to be recovered. There are also other CAD programs which read and write DWG's and will need the 99% information in these DWG's.

rurban, 2021-03-08:
There will never be 100% certainty. Even AutoCAD itself crashes on invalid input.

AutoCAD allows arbitrary 3rd party classes serialized to the Dwg. Teigha supports about 60% of them. We support about 40%, with some more common bugs. But no constraints and no parametric blocks yet. Not even proper acid solids or surfaces.
Simple dwg libs support about 10% and just skip the rest.

timoria21, 2021-03-08:
I did a test by myself, removing from JSON file all the problematic objects (most of them not related to geometry at all) and I was very close to get rid of the recovery mode error. Unfortunately, an issue with the one and only text style prevents me to succeed. Can you help me to understand what's wrong with this text style? Is there any text style data I can reset to avoid the problem?

[image]

All the documents attached.

[322.zip](https://github.com/LibreDWG/libredwg/files/6101630/322.zip)

rurban, 2021-03-08:
Still working on it. See GH #326

timoria21, 2021-03-11:
Can I expect a valid text style with your latest commit? Thanks.

timoria21, 2021-08-16:
Can this issue be related to x64 bitness, as your recent discovery on [this one](https://github.com/LibreDWG/libredwg/issues/346)? Thanks.

rurban, 2021-08-16:
Yes, I suspect a mingw64 problem with sizes

timoria21, 2023-03-28:
Hello,

Here is a detailed list of the steps that I'm doing in order to help you reproduce this issue on Linux.
You can find all the logs in the attached file [logs.zip](https://github.com/LibreDWG/libredwg/files/11087676/logs.zip).

I'm testing on a linux VM with Debian 11.

Some information on the OS, you can find the outputs of these commands inside info/:
apt list --installed -> /info/apt-list.txt
cat /etc/os-release -> /info/os-release.txt
hostnamectl -> /info/hostnamectl.txt

regarding apt-list, I installed specifically:
Build-Essential, Autoconf, Libtool, TeXinfo, GIT

From now on it's very straightforward, I just did the README.md steps one by one:

See INSTALL for generic instructions.  Basically, you do:
$ sh ./autogen.sh (if you checked out the source code from git)
$ ./configure [--enable-trace] [--disable-write] [--disable-shared]
$ make
$ make check
$ sudo make install

And again you can find all the respective outputs in the attached files (inside build/).

Finally, I tried to read a DWG file (Polyline3D.dwg, version AC1015 - ~2000) into a .json and then write it back to a DWG.
These are the commands that I've used (already tried with sudo too):

programs/dwgread -v9 -o ~/Desktop/dwgtest/Polyline3D.json ~/Desktop/dwgtest/Polyline3D.dwg 2>&1 | tee ~/Desktop/dwgtest/read_output.txt
programs/dwgwrite -v9 -o ~/Desktop/dwgtest/Polyline3DTEST.dwg ~/Desktop/dwgtest/Polyline3D.json 2>&1 | tee ~/Desktop/dwgtest/write_output.txt

But when I open "Polyline3DTEST.dwg" with autocad2007 I get the following message:

C:\Documents and Settings\Administrator\Desktop\Polyline3DTEST.dwg
Unable to open this drawing.
It contains incorrect or corrupted information.
The RECOVER command may be able to restore undamaged material from this drawing.

However, if I do recover the file everything seems fine.
You can find all the files and outputs inside dwg/.

What am I doing wrong?
If you want I can send you the VM that I'm using, VMware but I should be able to export it to other formats too.

Thank you.

rurban, 2023-03-28:
You are doing it right. AutoCAD just doesn't fully accept it yet

timoria21, 2023-03-28:
Ah, so we are back to [this issue](https://github.com/LibreDWG/libredwg/issues/346) again?

Thanks.

timoria21, 2023-03-30:
Thanks. Is "win64" still appropriate in the title of issue #346?

timoria21, 2023-03-30:
Do you mean I can try the latest changes already? Or do you still need to work on this?

I can repeat the test on Linux and let you know.

rurban, 2023-03-30:
Should work with the smoke/style-gh322 branch, yes.

timoria21, 2023-03-30:
Ok, will try tomorrow.

timoria21, 2023-04-07:
Can you repro? Do you need any other tests from my side?

Thanks.

timoria21, 2023-05-04:
Hello Reini,

So here we are stuck again, like two years ago...

Please let me know if I can do something else to help you with this issue: https://github.com/LibreDWG/libredwg/issues/322

Or even if a further donation would help.

Thank you again,

Dave

rurban, 2023-06-20:
Working on this in smoke/oda for a while.
Already fixed HATCH, VX_CONTROL, TABLECONTENT

problems still with TABLESTYLE, AcDbRasterImageDefReactor, AcDbMLeader
and some weird CRC error without any hint where

xsxWang, 2024-10-22:
> Automatic

how 

> Set the RECOVERAUTO variable

how to set the RECOVERAUTO variable?

tuxiasumari, 2026-08-29:
This no longer reproduces on 0.14.8583. Both reproducers in this thread now round-trip into DWGs that ODA File Converter reads without complaint.

Oracle and its limit, up front: I do not have AutoCAD, so I used **ODAFileConverter** as the judge — it rejects a malformed DWG with an `OdError` and writes a `.err` instead of the output. That is not literally "AutoCAD did not offer recovery", but it is the same class of structural check, and it does flag the old files, which is what makes the before/after meaningful.

Everything below is stock 0.14.8583, no local patches, source `Drawing2.dwg` / the `Polyline3D*` originals from this thread, target r2000.

| file | ODA |
|---|---|
| `Drawing2.dwg` (the AutoCAD original) | reads |
| **`Drawing2_C++.dwg`** (LibreDWG, 2021, attached above) | **rejects — `OdError: Null object Id: <object> (0)`** |
| `Drawing2.dwg` → dwg2dxf → dxf2dwg today | reads |
| **`Polyline3DR14TEST.dwg`, `Polyline3D04TEST.dwg`** (LibreDWG, 2023) | **rejects both** |
| the four `Polyline3D` originals (r14 / r2000 / r2004 / r2007) → round-tripped today | reads all four |

The rejections on the old outputs are the control: the oracle does catch the defect when it is there.

And it is not an empty pass. Comparing the ODA-converted original against the ODA-converted round-trip, per source version:

```
r14  : 163 LWPOLYLINE, 30925 points   ->  163 LWPOLYLINE, 30925 points
r2000: 163 LWPOLYLINE, 30925 points   ->  163 LWPOLYLINE, 30925 points
r2004: 163 LWPOLYLINE, 30925 points   ->  163 LWPOLYLINE, 30925 points
r2007: 163 LWPOLYLINE, 30925 points   ->  163 LWPOLYLINE, 30925 points
```

X range identical to the millimetre (-322860.176 .. 715515.875) in all four.

Three real drawings of mine behave the same way through the write path: 9847 -> 9847 modelspace entities and 744 -> 744 paperspace entities on one of them, no entity type lost.

`Null object Id` is a dangling handle reference, which is the family #1370 was in (`in_postprocess_SEQEND` rewrapping a relative handle as absolute made AutoCAD reject the whole DWG). Several of the handle/ownership fixes that landed since 0.14 plausibly cover this.

Suggest closing, or reopening with a current reproducer if someone still sees recovery mode on a file written by 0.14.8583+.