LibreDWG/libredwg #325: Heap-buffer-overflow in decode.c:470

zodf0055980, 2021-03-04:
I found a heap buffer overflow in the current master (8072563).

# Configure
CFLAGS="-g -fsanitize=address" LDFLAGS="-fsanitize=address" ./configure

# Command
`./dwglayers ./poc`

# ASAN report
```
➜  ./dwglayers ./poc
=================================================================
==23168==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x6120000001c8 at pc 0x7ffff29b78be bp 0x7fffffffa9c0 sp 0x7fffffffa9b0
WRITE of size 8 at 0x6120000001c8 thread T0
    #0 0x7ffff29b78bd in decode_preR13_section /home/yuan/afl-target/libredwg-asan/src/decode.c:470
    #1 0x7ffff2a26e90 in decode_preR13 /home/yuan/afl-target/libredwg-asan/src/decode.c:744
    #2 0x7ffff29a43e9 in dwg_decode /home/yuan/afl-target/libredwg-asan/src/decode.c:235
    #3 0x7ffff297afa4 in dwg_read_file /home/yuan/afl-target/libredwg-asan/src/dwg.c:253
    #4 0x55555555688e in main /home/yuan/afl-target/libredwg-asan/programs/dwglayers.c:148
    #5 0x7ffff1f52bf6 in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x21bf6)
    #6 0x555555555bf9 in _start (/home/yuan/afl-target/libredwg-asan/programs/.libs/dwglayers+0x1bf9)

Address 0x6120000001c8 is a wild pointer.
SUMMARY: AddressSanitizer: heap-buffer-overflow /home/yuan/afl-target/libredwg-asan/src/decode.c:470 in decode_preR13_section
Shadow bytes around the buggy address:
  0x0c247fff7fe0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c247fff7ff0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c247fff8000: fa fa fa fa fa fa fa fa 00 00 00 00 00 00 00 00
  0x0c247fff8010: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c247fff8020: 00 00 00 00 00 00 00 00 00 00 00 00 00 fa fa fa
=>0x0c247fff8030: fa fa fa fa fa fa fa fa fa[fa]fa fa fa fa fa fa
  0x0c247fff8040: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c247fff8050: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c247fff8060: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c247fff8070: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c247fff8080: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07 
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
==23168==ABORTING

```
# POC
[poc.zip](https://github.com/LibreDWG/libredwg/files/6080842/poc.zip)

rurban, 2021-03-06:
We don't release this code-path yet. fuzzers should configure with `--enable-release`. it is only run with git checkouts.

Fixed with d44b47bbeceb2365c66a57084a1fdcfb846126dc

zodf0055980, 2021-03-06:
Sorry, I will add `--enable-release` to run fuzzing, thanks.

closed 2021-03-08 by rurban