LibreDWG/libredwg #326: Version 12.2 loads this DWG much worse compared to v12.1

timoria21, 2021-03-04:
Hello,

Trying the attached DWG file with the latest build we get a much worse result. Screenshots are included in the attached zip file.

[TestDWG.zip](https://github.com/LibreDWG/libredwg/files/6081593/TestDWG.zip)

rurban, 2021-03-06:
Yes, should be fixed with 8e665f0273de973cbf2c804d999be5d0f788737d

More bitsize and size recalculations are needed with DD types, if missing leading to wrong hdl offsets.

rurban, 2021-03-08:
Nope, still wrong sizes

rurban, 2021-03-08:
with 0.12.1 we got 59996 rewrite errors with this testcase.
with 14527d3eed139a23a6cd165ed591be8337ad18ba we get 187 errors.
with 7d4210cff1d34d226b8eacd4e4b06a418ef14a54 we get 63 errors.

rurban, 2021-03-09:
what is remaining is a wrong scale DD encode for scale.z 3 vs 0

timoria21, 2021-03-19:
I tried to build and run libredwg-0.12.3.4231-win64 but the issue is still there (see attached files). 
[image]
[TestDWG_12.3.zip](https://github.com/LibreDWG/libredwg/files/6170069/TestDWG_12.3.zip)

What am I missing?

rurban, 2021-03-19:
Thanks for the recovery log. Looks like the BLOCK "1573" is broken, and more. 
I suspect BLOCK subentity and POLYLINE vertex ordering a bit broken...

timoria21, 2021-04-07:
Hello, any news on this topic?

rurban, 2021-04-07:
Yes, see above. Not repro