LibreDWG/libredwg #327: Certain texts are not shown (and other issues) after dwg to dxf conversion

plaes, 2021-04-06:
I have a dwg drawing and after converting it to dxf, dimensions are not visible in the result (when trying out with both QCAD and LibreCAD:
[image]

Same file opened in the online dwgfastview application:
[image]

[cellar_plan.zip](https://github.com/LibreDWG/libredwg/files/6264815/cellar_plan.zip)

plaes, 2021-04-06:
Ok, apparently it's the issue of the famous unit conversions.. :)

rurban, 2021-04-06:
Those texts are stored as "<>" in both the DWG and the DXF. I'll have to check of the DXF needs the computed values there instead.