LibreDWG/libredwg #332: Any expectation for dwgwrite from GeoJSON

jossefaz, 2021-04-08:
All is in the title : is there any time expectation for this feature ?

Best Regards

rurban, 2021-04-08:
Oh, no idea. Is this needed? Easier would be via DXF, via some script.

jossefaz, 2021-04-08:
I successfully convert a GeoJSON to DXF with the [GDAL](https://gdal.org/) `ogr2ogr` command line.

But then I tried both `dwgwrite` and `dxf2dwg` command and both did not output a valid DWG file.

Here is a link to the [DXF](https://drive.google.com/file/d/155JgGfI16RxHXuToDZztpth6zOoqkjud/view?usp=sharing) output from GDAL. If it could help anyhow.

rurban, 2021-04-08:
Great that GDAL handles that already. So we only have to understand the GDAL DXF, which should be easier. Thanks, will fix that

rurban, 2021-04-09:
ogr2ogr converts a POLYLINE to a HATCH? interesting. I would have expected a POLYLINE_2D

jossefaz, 2021-04-11:
Well as far as I know there is no Polygon entity natively in AutoCad (I use AutoCad Map btw...it might be a slight difference on what I write here with AutoCAD 3D).

And the input GeoJSON of that conversion has Polygons features.
This is why you can see the same XY coordinates at the beginins and at the end of the HATCH. 
I assume that if my GeoJSON has instead a LineString feature, then it should have been converted to POLYLINE_2D  

Hope this helps...

rurban, 2021-04-11:
But what's your GDAL version? It might be something earlier than 3.0.4, because I can successfully import my 3.0.4 DXF's, but not yours. Or it's some feature I have to find.

jossefaz, 2021-04-11:
I use the official docker image of [Osgeo](https://github.com/OSGeo/gdal/tree/master/gdal/docker)
The version there is 
`ogr2ogr --version  GDAL 3.3.0dev-7856bfcd6e516a93d7cb23da70a13c41dc4214c4, released 2021/04/04`

rurban, 2021-04-12:
I'd need your geojson file also. with my examples cannot reproduce it. Even latest GDAL from git master works fine.

mumumi, 2024-10-15:
@rurban here it is
[test.zip](https://github.com/user-attachments/files/17373345/test.zip)

rurban, 2025-06-16:
- fromGDAL.dxf converts and loads fine.
- output.dxf from test.zip also converts and loads fine.

pcace, 2025-06-16:
thats not the case for me: 

this is my geojson: 
(made it a linestring to avoid ogr2ogr to create a hatch instead of a polygon)

```
{
    "type": "FeatureCollection",
    "features": [
        {
            "type": "Feature",
            "properties": {},
            "geometry": {
                "type": "LineString",
                "coordinates": 
                    [
                        [
                            1,
                            1
                        ],
                        [
                            1,
                            2
                        ],
                        [
                            2,
                            2
                        ],
                        [
                            2,
                            1
                        ],
                        [
                            1,
                            1
                        
                    ]
                ]
            }
        }
    ]
}

```
this is my dxf conversion:

```
ogr2ogr -f "DXF" output.dxf test.json -lco DXF_WRITE_HATCH=FALSE  -lco ENCODING=UTF-8   -lco HEADER=NO
```

i then create a dwg with dxf2dwg: 

```
dxf2dwg output.dxf 
Reading DXF file output.dxf
Warning: Unknown HEADER.CMLSTYLE Standard dxf:2
Warning: Unknown HEADER.CMLSTYLE Standard dxf:2
Warning: Object handle not found 131073/0x20001
Warning: Object handle not found 131074/0x20002
Writing DWG file output.dwg

```

i can open the dxf in any CAD application, but the dwg is empty.

[outputdxf.zip](https://github.com/user-attachments/files/20756072/outputdxf.zip)
[outputdwg.zip](https://github.com/user-attachments/files/20756071/outputdwg.zip)