LibreDWG/libredwg #333: CMake Erro

wangxw0820, 2021-04-09:
I got a error When I am using cmake：

CMake Error at CMakeLists.txt:226 (install):
  install TARGETS given no LIBRARY DESTINATION for shared library target
  "redwg".

rurban, 2021-04-09:
Does this fix it?
```patch
diff --git CMakeLists.txt CMakeLists.txt
index 952453c39..ef06c929d 100644
--- CMakeLists.txt
+++ CMakeLists.txt
@@ -226,7 +226,7 @@ endif()
 install(TARGETS redwg
   LIBRARY
     COMPONENT redwg
-    )
+  DESTINATION libredwg-${PACKAGE_VERSION})
 install(TARGETS RUNTIME)
 
 if(MSVC)
```

closed 2021-04-09 by wangxw0820

j3pic, 2022-11-14:
This works, but why is this bug closed? I just got this error, and it's clearly not fixed in the repo.

rurban, 2022-11-14:
Oops, got lost apparently. Just pushed it to master