LibreDWG/libredwg #335: I'm new to the project, could you please tell me how to read the UserCoordinateSystem of a dwg?

ChengTheRipper, 2021-04-16:
I get the UCS_Control from the dwg_data and dont know what to do next.
I'm really looking forward to some help.

ChengTheRipper, 2021-04-16:
I try to output the ucs_control's num entries, and it said zero.
However I do set one...

rurban, 2021-04-17:
You really need to study AutoCAD docs for this.

Each entity UCS is defined by its ECS (Entity Coord. System), defined by its normal vector, the 210 extrusion. From there is a right hand rule to get the ECS. The UCS entries just store the this normal under 3 other vectors ucsorg (the new 0,0), and ucsxdir, ucsydir under a name.

ChengTheRipper, 2021-04-17:
I see. Thank you very much.
 I'll take a look at the AutoCAD doc to get the general idea of your advice.
I really dont know how to get my hands on the project, except for reading the examples.
I thought the UCS is store in a table object, and I can read it through table object.
Once again. Thank you very much.

closed 2021-04-17 by ChengTheRipper