LibreDWG/libredwg #337: Using dwggrep to find all distinct geometry types (or subclasses) in dwg

YaroslavKormushyn, 2021-04-21:
Hello!

I've found that `dwgfilter` is no longer in the program list and figured that `dwggrep` should be a replacement for it, then.

How would one query all geometry types in a dwg file e.g. Polyline, Polygon, MText etc.?

YaroslavKormushyn, 2021-04-21:
There's a [section in the documentation](https://www.gnu.org/software/libredwg/manual/LibreDWG.html#Programs) for `dwgfilter`, but I don't see that program available in the Windows x64 binaries.

YaroslavKormushyn, 2021-04-21:
I've tried opening the `dwggrep.exe` and had missing dlls pop up. I believe you already fixed it in #103, but I don't see the dlls included anywhere in the zip of the [latest release](https://github.com/LibreDWG/libredwg/releases/tag/0.12.4).

<summary> Screenshots <details>

[image]

[image]

</details>
</summary>

YaroslavKormushyn, 2021-04-22:
1. So that means `dwgfilter` is only going to be available for machines running Linux?
2. Ok, that makes sense, but I thought they're going to be included in the package, since the other dependencies are there...

rurban, 2021-04-25:
Add dwgfilter to the windows zip. Maybe find a bash => bat converter

closed 2021-05-14 by rurban