LibreDWG/libredwg #342: dxf.test fails, roundtrip fail: expect 259, got 262 entities

AndrewAmmerlaan, 2021-05-24:
Hello,

Running `make check` fails for me on a Gentoo Linux system with version 0.12.4.

```
PASS: alive.test
Reading DWG file ../test/test-data/2000/PolyLine2D.dwg
Writing DXF file ../PolyLine2D_2000.dxf
Reading DWG file ../test/test-data/example_r14.dwg
Writing DXF file ../example_r14.dxf
Reading DWG file ../test/test-data/example_2000.dwg
Writing DXF file ../example_2000.dxf
Reading DWG file ../test/test-data/example_2004.dwg
Writing DXF file ../example_2004.dxf
Reading DWG file ../test/test-data/example_2007.dwg
Writing DXF file ../example_2007.dxf
Reading DWG file ../test/test-data/example_2010.dwg
Writing DXF file ../example_2010.dxf
Reading DWG file ../test/test-data/example_2013.dwg
Writing DXF file ../example_2013.dxf
Reading DWG file ../test/test-data/example_2018.dwg
Writing DXF file ../example_2018.dxf
../sample_2000.dxf ACDBDICTIONARYWDFLT ok
../sample_2000.dxf ACDBPLACEHOLDER ok
../sample_2000.dxf LAYOUT ok
../sample_2018.dxf ACDBDICTIONARYWDFLT ok
../sample_2018.dxf ACDBPLACEHOLDER ok
../sample_2018.dxf LAYOUT ok
../example_2000.dxf ACDBPERSSUBENTMANAGER ok
../example_2000.dxf ACDBASSOCDEPENDENCY ok
../example_2004.dxf ACDBPERSSUBENTMANAGER ok
../example_2004.dxf ACDBASSOCDEPENDENCY ok
../example_2007.dxf ACDBPERSSUBENTMANAGER ok
../example_2007.dxf ACDBASSOCDEPENDENCY ok
../example_2010.dxf ACDBPERSSUBENTMANAGER ok
../example_2010.dxf ACDBASSOCDEPENDENCY ok
../example_2013.dxf ACDBPERSSUBENTMANAGER ok
../example_2013.dxf ACDBASSOCDEPENDENCY ok
../example_2018.dxf ACDBPERSSUBENTMANAGER ok
../example_2018.dxf ACDBASSOCDEPENDENCY ok
../Underlay_2004.dxf PDFDEFINITION ok
../Underlay_2004.dxf PDFUNDERLAY ok
../example_2000.dxf pline order ok
../example_2004.dxf pline order ok
../example_2007.dxf pline order ok
../example_2010.dxf pline order ok
../example_2013.dxf pline order ok
../example_2018.dxf pline order ok
../example_2000.dxf insert order ok
../example_2004.dxf insert order ok
../example_2007.dxf insert order ok
../example_2010.dxf insert order ok
../example_2013.dxf insert order ok
../example_2018.dxf insert order ok
example_2000.dxf roundtrip fail: expect 259, got 262 entities
--- example_2000.log1.tmp       2021-05-24 08:19:52.247668390 -0000
+++ example_2000.log2.tmp       2021-05-24 08:19:52.250668354 -0000
-WIPEOUT
+POINT
-WIPEOUT
+POINT
example_2004.dxf roundtrip fail: expect 259, got 262 entities
--- example_2004.log1.tmp       2021-05-24 08:19:53.026659145 -0000
+++ example_2004.log2.tmp       2021-05-24 08:19:53.038659003 -0000
-WIPEOUT
+POINT
-WIPEOUT
+POINT
example_2007.dxf roundtrip fail: expect 259, got 262 entities
--- example_2007.log1.tmp       2021-05-24 08:19:54.098646424 -0000
+++ example_2007.log2.tmp       2021-05-24 08:19:54.107646317 -0000
-WIPEOUT
+POINT
-WIPEOUT
+POINT
example_2010.dxf roundtrip fail: expect 259, got 262 entities
--- example_2010.log1.tmp       2021-05-24 08:19:55.147633978 -0000
+++ example_2010.log2.tmp       2021-05-24 08:19:55.154633894 -0000
+TABLE
-WIPEOUT
+POINT
-TABLE
-WIPEOUT
+POINT
example_2013.dxf roundtrip fail: expect 259, got 262 entities
--- example_2013.log1.tmp       2021-05-24 08:19:56.048623288 -0000
+++ example_2013.log2.tmp       2021-05-24 08:19:56.052623241 -0000
+TABLE
-WIPEOUT
+POINT
-TABLE
-WIPEOUT
+POINT
example_2018.dxf roundtrip fail: expect 259, got 262 entities
--- example_2018.log1.tmp       2021-05-24 08:19:57.011611864 -0000
+++ example_2018.log2.tmp       2021-05-24 08:19:57.015611816 -0000
+TABLE
-WIPEOUT
+POINT
-TABLE
-WIPEOUT
+POINT
dxf.test: 6 failures
ls: cannot access './*dxf.log': No such file or directory
FAIL: dxf.test
=================================
1 of 2 tests failed
Please report to libredwg@gnu.org
=================================
```

Please let me know if you need any additional information.

rurban, 2021-05-25:
Yes, occasionally I get that also. It's a wrong test.
In this case several unstable entities are converted to POINTs with EED attached.

rurban, 2024-02-06:
Fixed last year

AndrewAmmerlaan, 2024-02-06:
Great, Thanks!

The orignal failures I saw are indeed gone in the latest release. I'm now seeing a new failure though:
```
example_r13.dxf roundtrip fail: expect 5287, got 5286 entities
--- example_r13.log1.tmp        2024-02-06 12:20:11.150654329 +0100
+++ example_r13.log2.tmp        2024-02-06 12:20:11.163987665 +0100
-WIPEOUT
-ARC_DIMENSION
-TABLE
-WIPEOUT
-LIGHT
+MINSERT
dxf.test: 1 failures
ls: cannot access './*dxf.log': No such file or directory
FAIL: dxf.test
=================================
1 of 2 tests failed
Please report to libredwg@gnu.org
=================================
```

rurban, 2024-02-06:
Hmm, another test failure. MINSERT are wrong for r13, and therefore skipped

closed 2024-02-26 by rurban