LibreDWG/libredwg #346: or “Standard” style the “Font name” and “Font style” fields are coming blank on text style dialog

Dev7741, 2021-06-03:


Dev7741, 2021-06-03:
After opening DWG in AutoCAD 2013 version for "Text style" dialog "Font Name" and "Font Style" field coming as blank for "Standard" style. Please refer below image.

[image]

rurban, 2021-06-03:
But how did you create this DWG exactly? DXF, Json, dwgadd? Can I have the source of this?

Dev7741, 2021-06-06:
I have created DWG using Json file on latest release build "0.12.4.4317". Please find attached Zip file contain 
JSON file.

[Circle_Version_2000.zip](https://github.com/LibreDWG/libredwg/files/6603567/Circle_Version_2000.zip)

Dev7741, 2021-06-14:
@rurban 
Are you able to reproduce the issue?

rurban, 2021-06-14:
No, no time yet, sorry

Dev7741, 2021-07-20:
@rurban 
Any updates?

timoria21, 2021-07-22:
Hello, 

I'm ready to donate $100 for getting all the details to resolve this issue? Can somebody provide me the donation link?

Thanks.

rurban, 2021-08-09:
Not able to reproduce with AutoCAD r2018. 
See my style dialog
[image]
after converting your json to a r2000 DWG with current master.
The AutoCAD Importer also reports no STYLE error.

Dev7741, 2021-08-13:
Hi,

I have checked on latest release build (0.12.4.4348) using same JSON file on AutoCAD 2013 and AutoCAD 2018(windows 10- 64 bit) however issue is reproducible on my machine.
please see below snapshot.
[image]

Also reported text Style error. Please see below Error.

Reading handle 11 object type AcDbTextStyleTableRecord
      Error 34 (eWrongObjectType)                       Object discarded
Reading handle 132 object type AcDbTextStyleTableRecord
      Error 34 (eWrongObjectType)                       Object discarded

on my machine Arial.ttf font file is at default location i.e. "C:\Windows\Fonts". Do we need to copy font file to other location?

rurban, 2021-08-13:
This seems to be the problem:
```
Reading handle 11 object type AcDbTextStyleTableRecord
Error 34 (eWrongObjectType) Object discarded
Reading handle 132 object type AcDbTextStyleTableRecord
Error 34 (eWrongObjectType) Object discarded
```

rurban, 2021-08-14:
Can you also attach the generated DWG and LOG file, please. Must be some windows thing then, I think

Dev7741, 2021-08-15:
Please find attached SampleFiles.zip folder
[SampleFiles.zip](https://github.com/LibreDWG/libredwg/files/6987691/SampleFiles.zip)
.
It contains:
Original DWG file name: Circle_Original.dwg
JSON file name: Circle_Test_2000.json
Generated DWG file name: Circle_Newely_generate.dwg
Log file name: LogFile.txt 

Please let me know if more information is required.

rurban, 2021-08-15:
I can repro it on windows. Some wrong size calculation!

rurban, 2021-09-07:
no sorry. I was on holidays, and haven't fixed my windows vm/machine yet

timoria21, 2021-09-16:
Reini,

If you tell me what to check I can try by myself.

rurban, 2021-09-16:
Debug into the handle corruption

Dev7741, 2021-09-24:
Did you find any handle corruption code. If you know any class or function where there is chance of handle corruption. Please let us know we will debug the particular code.

timoria21, 2021-09-28:
Hi Reini,

Can you please be more specific? Where in the code do I need to check?

Thank you.

rurban, 2021-09-30:
Fixed one part.
remaining error:
`make && programs/dwgwrite.exe -v1 -y Circle_Version_2000.json`
=> Warning: Wrong object size: 19659 + 18 = 19679 != 19691: -12 off ...

rurban, 2021-10-01:
Fixed the 2nd part, no more corrupt handle sizes.

Dev7741, 2021-10-04:
Thanks for the fix.

We have merged the modified code at our end and observed that out of two error one is fixed and below error is reproducible at our end. 
**Reading handle 11 object type AcDbTextStyleTableRecord
Error 34 (eWrongObjectType) Object discarded**

If possible could you please provide latest release build so we can check at our end.

rurban, 2021-10-04:
Still working on the merge. mingw is broken a bit

timoria21, 2021-10-05:
Thanks Reini, I feel we are very close to resolving the issue.

rurban, 2021-10-05:
Is it good now?

timoria21, 2021-10-05:
No, please go on with the merge.

timoria21, 2021-10-05:
Why the issue is now tagged "need info"? I was simply happy to see that you were doing progress on this.

rurban, 2021-10-05:
David J. Harris ***@***.***> schrieb am Di., 5. Okt. 2021,
20:20:

> Why the issue is now tagged "need info"? I was simply happy to see that
> you were doing progress on this.
>

Because i need feedback if the problem is now solved or not.

Dev7741, 2021-10-06:
Hi,

I have checked on latest release build (0.12.4.4362) using same JSON file (Circle_Test_2000.json) on AutoCAD 2013(windows 10- 64 bit) however newly created drawing get crashed while opening drawing. Please see below message get displayed while opening drawing.

[image]

Please let us know if more information is required.
Thanks

timoria21, 2021-11-08:
Hi Reini,

I finally discovered where the problem is, the fix is valid for AutoCAD v2007 and v2012 and **not** for AutoCAD v2013 and up. 

In particular, I've tested v2013, DwgTrueView 2014 and DwgTrueView v2021.

Can you please check?

Thanks.

timoria21, 2021-11-18:
Hi again. Is there anything I can check by myself?

rurban, 2021-11-18:
> Hi again. Is there anything I can check by myself?
If you can.

timoria21, 2021-11-19:
Sorry, I don't understand your reply. If you provide me with some context and the things I need to check I can try to help you. Without any instruction, I don't know where to start... Thanks again.

timoria21, 2022-01-09:
Happy New Year Reini,

I'm close to giving up my project based on LibreDWG because of the lack of support of the latest version of AutoCAD for the issue: https://github.com/LibreDWG/libredwg/issues/346

Unfortunately, without your help, I am unable to resolve this problem by myself and even after offering you a new donation you don't reply anymore.

You are probably busy on different projects. I want to thank you for the effort and please confirm you are unable/uninterested to help me with this.

Thanks again,

Dave
[https://opengraph.githubassets.com/66e51f7bfd15175428ea452f8b0dbab08e1d2df207df6221b86e0b1c5277f967/LibreDWG/libredwg/issues/346]<https://github.com/LibreDWG/libredwg/issues/346>
Broken Text Style write on win64 only · Issue #346 · LibreDWG/libredwg<https://github.com/LibreDWG/libredwg/issues/346>
With font_file: arial.ttf (EED name "Arial", flag 34).
github.com

rurban, 2022-01-09:
I'm interested, but that's a very hard problem for me, when I cannot reproduce it and have no further ideas. This takes time.

timoria21, 2022-01-11:
Do you mean you don't have AutoCAD 2013 or up to reproduce the issue?

rurban, 2022-01-11:
On Tue, Jan 11, 2022 at 9:51 AM David J. Harris ***@***.***>
wrote:

> Do you mean you don't have AutoCAD 2013 or up to reproduce the issue?
>

I do have an AutoCAD 2018, but on mac, where I cannot reproduce it.
I don't have a Windows PC currently, this is in work.

timoria21, 2022-01-14:
Please give me the exact instructions on what you would do on AutoCAD 2018 on Windows PC to make your tests and I will be more than happy to do them for you and attach the results here.

timoria21, 2022-01-19:
Hello, what if I give you access to my Windows machine using TeamViewer with an AutoCAD 2013+ version? Did you consider installing AutoCAD on Parallels on your Mac?

timoria21, 2022-03-02:
Hello again, any update on the AutoCAD 2018 on Windows PC ?

Thank you.

timoria21, 2022-08-14:
Hello,

I've just donated another $100 for this problem to be resolved.

More than one year is passed since my last donation, I hope you can find some time to fix it.

Dave

timoria21, 2022-12-28:
Hello Reini,

Any chance to resolve this issue? Would another donation help?

Thanks, 

Dave

rurban, 2023-03-28:
Almost: 
ad 4) AutoCAD can open it, but goes into recovery mode with an error in the textstyle. There seems to be some undocumented slack. See #322

michal-josef-spacek, 2023-07-18:
@timoria21 Could you respond to my question in https://github.com/LibreDWG/libredwg/issues/346#issuecomment-1486857584 ? 

btw: There was progress in fixing bugs in conversion from DWG to JSON in LibreDWG. In the case of your file, there are still some issues.
I created https://metacpan.org/pod/test-dwg-libredwg-json tool, which converts DWG file to JSON, back to DWG and back to JSON, and diff JSONs. Actual result is:
```
697c697
<       "strings_area": "00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000"
---
>       "strings_area": "30000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000"
716c716
<       "strings_area": "00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000"
---
>       "strings_area": "30000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000"
735c735
<       "strings_area": "00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000"
---
>       "strings_area": "30000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000"
1188c1188
<       "unknown_bits": "4254DD185B99185C990028000000000000F83F0000000000003E0FC00000000000004901028000000C1008070000C3001FDFF8000000C1003FBFF000000182007F7FE00000030400FEFFC00000060801FDFF8000000C1003FBFF000000182000000000000000C2020B0000001820100E000186003FBFF000000182007F7FE00000030400FEFFC00000060801FDFF8000000C1003FBFF000000182007F7FE0000003040000000000000012404160000003040201C00030C007F7FE00000030400FEFFC00000060801FDFF8000000C1003FBFF000000182007F7FE00000030400FEFFC00000060804020BF1900B12888A888A88803",
---
>       "unknown_bits": "4254DD185B99185C990028000000000000F83F0000000000003E0FC00000000000004901028000000C1008000000C0001FDFF8000000C1003FBFF000000182007F7FE00000030400FEFFC00000060801FDFF8000000C1003FBFF000000182000000000000000C2020B00000018201000000180003FBFF000000182007F7FE00000030400FEFFC00000060801FDFF8000000C1003FBFF000000182007F7FE00000030400000000000000124041600000030402000000300007F7FE00000030400FEFFC00000060801FDFF8000000C1003FBFF000000182007F7FE00000030400FEFFC00000060804020BF1900B12888A888A88803",
1206,1207c1206
<             "index": 256,
<             "rgb": "c3000007"
---
>             "rgb": "c0000000"
1263,1264c1262
<             "index": 256,
<             "rgb": "c3000007"
---
>             "rgb": "c0000000"
1320,1321c1318
<             "index": 256,
<             "rgb": "c3000007"
---
>             "rgb": "c0000000"
5711c5708
<             "rgb": "c8000000"
---
>             "rgb": "c1000000"
5819c5816
<             "rgb": "c8000000"
---
>             "rgb": "c1000000"
5927c5924
<             "rgb": "c8000000"
---
>             "rgb": "c1000000"
```
This means, that are structural issues in the DWG file from JSON conversion.

michal-josef-spacek, 2023-07-18:
@timoria21 
When I ```dwgrewrite``` your original file, there are still some issues in AutoCAD 2000. 
And the difference between DWG files are same as diff in the previous comment. I believe that we need to fix these issues first.

timoria21, 2023-07-20:
Thanks for checking this for me, again let me know if I can help/test in some way.

So, nobody has an idea of what triggers the RecoveryMode in AutoCAD.

I think it would be a great value for the LibreDWG project to write DWG files correctly.

michal-josef-spacek, 2023-07-21:
After fixing of strings_area issue still persists.

michal-josef-spacek, 2023-07-27:
After the provisional fix of https://github.com/LibreDWG/libredwg/issues/785 I have an equal JSON result.
Interesting.
Issue still persists.

So I am going to compare binary files.

michal-josef-spacek, 2023-07-31:
After the provisional fix of #788 issue still persists.

michal-josef-spacek, 2023-08-15:
@timoria21 I believe that the main issue of the DWG file which needs to be recovered is with a bad second header in conversion from JSON to DWG. After fixing https://github.com/LibreDWG/libredwg/issues/792 we need to check.

rurban, 2023-08-16:
the remaining errors are in the CI, see eg. https://github.com/LibreDWG/libredwg/actions/runs/5870138178/job/15916583698

example_2018.dwg
  Object of class AcDbLayerTableRecord can't be cast to AcDbEntity.

earlier only some unknown recovery errors.

michal-josef-spacek, 2023-08-17:
hm, after fixing of previous, the issue is still here.

michal-josef-spacek, 2023-08-17:
> the remaining errors are in the CI under **oda** see eg. https://github.com/LibreDWG/libredwg/actions/runs/5870138178/job/15916583698
> 
> example_2018.dwg Object of class AcDbLayerTableRecord can't be cast to AcDbEntity. example_r14.dwg CRC does not match
> 
> earlier only some unknown recovery errors.

But this is roundtripping over the DXF file. Conversion to and from DXF is more problematic than JSON now.

michal-josef-spacek, 2023-08-17:
I am testing roundtripping over the JSON and I have a simple line in AC1015, which is not working (need to be recovered).
The main difference is in texts and then in sizes because of different texts.
e.g.  header
```
unit1_name: "m" [TV 0]
```
vs
```
TV-ZERO 2
unit1_name: "m" [TV 0]
```

I have a blank file, which has the same issues, but it's working after conversion without recovery.
I don't see the root cause.

The same issue with recovery is on R14 file with one line.

michal-josef-spacek, 2023-08-17:
The same issue is in the AC1012 file. Testing is blocked by https://github.com/LibreDWG/libredwg/issues/805 issue now.

rurban, 2023-08-21:
Circle_Original.dwg can now properly json roundtrip without any teigha error anymore

michal-josef-spacek, 2023-08-21:
JSON roundtrip by ```test-dwg-libredwg-json Circle_Original.dwg```
```
33091,33092c33091,33092
<     "size": 117,
<     "address": 163713,
---
>     "size": 108,
>     "address": 164204,
33120,33121c33120,33121
<         "address": 163035,
<         "size": 678
---
>         "address": 163037,
>         "size": 1167
33125,33126c33125,33126
<         "address": 0,
<         "size": 0
---
>         "address": 164204,
>         "size": 117
33227c33227
<     "crc": 57194
---
>     "crc": 41100
```
There are only second header changes

The issue with recovery mode is present still.

There is something very wrong. e.g. The output of ```dwgadd``` for many AutoCAD versions is not usable by recovery issue.
In AC1012-AC1015 maybe because write support of Thumbnail.

timoria21, 2023-08-29:
_I had a situation when the output of DWG->JSON->DWG was without recovery mode but without 2nd header. I believe, that 2nd header is the important thing in the DWG structure._

Is this something we can remove, to get rid of the "recovery mode"?

rurban, 2023-08-29:
> _I had a situation when the output of DWG->JSON->DWG was without recovery mode but without 2nd header. I believe, that 2nd header is the important thing in the DWG structure._
> 
> Is this something we can remove, to get rid of the "recovery mode"?

No, that's already fixed. Now we are getting closer. The remaining problems are some downconverting EED's mostly. The list of roundtrip errors is getting shorter every day

timoria21, 2023-08-29:
Wow, thanks! I look forward for this fix.

michal-josef-spacek, 2023-08-29:
> The list of roundtrip errors is getting shorter every day

@rurban Do you have some roundtrip errors? How to reproduce?

[71 more replies]

closed 2024-06-14 by rurban