LibreDWG/libredwg #348: heap-overflow in function read_data_section in decode_r2007.c

chibataiki, 2021-06-05:
Hi,
In release version 0.12.4, (also in the latest commit 4c210bb)

There is overflow in function read_data_section in decode_r2007.c

This bug is caused by the following dangerous memcpy calling
```c
// decode_r2007.c
821	      else
822	        {
823	          memcpy (&decomp[section_page->offset], &dat->chain[dat->byte],
824	                  section_page->uncomp_size);
825	        }
826	    }
827	  sec_dat->chain = decomp;
──── threads ────
[#0] Id 1, Name: "dwg2SVG", stopped 0x7ffff7d7595e in __memmove_avx_unaligned_erms (), reason: SIGSEGV
─── trace ────
[#0] 0x7ffff7d7595e → __memmove_avx_unaligned_erms()
[#1] 0x555555d10792 → read_data_section(sec_dat=0x555556988e20 <sec_dat>, dat=0x7fffffffd090, sections_map=0x55555698d070, pages_map=0x55555698d740, sec_type=SECTION_PREVIEW)
[#2] 0x555555d39987 → read_2007_section_preview(dat=0x7fffffffd090, dwg=0x7fffffffd1a0, sections_map=0x55555698d070, pages_map=0x55555698d740)
[#3] 0x555555d3a268 → read_r2007_meta_data(dat=0x7fffffffd090, hdl_dat=0x7fffffffcfd0, dwg=0x7fffffffd1a0)
[#4] 0x555555634d2a → decode_R2007(dat=0x7fffffffd090, dwg=0x7fffffffd1a0)
[#5] 0x5555555bedba → dwg_decode(dat=0x7fffffffd090, dwg=0x7fffffffd1a0)
[#6] 0x5555555adfc0 → dwg_read_file(filename=0x7fffffffe611 "poc", dwg=0x7fffffffd1a0)
[#7] 0x5555555acc65 → main(argc=0x2, argv=0x7fffffffe378)

```
./configure --enable-release
make
./bin/dwg2SVG [poc]

[heapoverflow_read_data_section_1.zip](https://github.com/LibreDWG/libredwg/files/6603028/heapoverflow_read_data_section_1.zip)

giantbranch, 2021-06-08:
@rurban heap overflow read still exists.

rurban, 2021-06-08:
Oh, I'll check in the evening

rurban, 2022-02-06:
Fixed in 0.12.5

closed 2022-02-06 by rurban