LibreDWG/libredwg #349: stack-buffer-overflow in function dwg_section_wtype in dwg.c.

chibataiki, 2021-06-05:
Hi,
In release version 0.12.4, (also in the latest commit 4c210bb)

There is stack-buffer-overflow in function dwg_section_wtype in dwg.c

Size of array `name` is 24 ,but in the while loop at line 1661 , `name` will overflow if `i` is big enough

```c
//dwg.c
1654      char name[24];
...
1661      while ((c = *wp++))
1662        {
1663          name[i++] = (char)(c & 0xff);
1664        }

──────── threads ────
[#0] Id 1, Name: "dwgread", stopped 0x7ffff7cbd18b in __GI_raise (), reason: SIGABRT
───────── trace ────
[#0] 0x7ffff7cbd18b → __GI_raise(sig=0x6)
[#1] 0x7ffff7c9c859 → __GI_abort()
[#2] 0x7ffff7d073ee → __libc_message(action=do_abort, fmt=0x7ffff7e3107c "*** %s ***: terminated\n")
[#3] 0x7ffff7da9b4a → __GI___fortify_fail(msg=0x7ffff7e31064 "stack smashing detected")
[#4] 0x7ffff7da9b16 → __stack_chk_fail()
[#5] 0x5555555b003d → dwg_section_wtype(wname=0x5555573491e0)
[#6] 0x555555d0ef79 → read_sections_map(dat=0x7fffffffd040, size_comp=0x47f, size_uncomp=0xa14, correction=0x4)
[#7] 0x555555d38230 → read_r2007_meta_data(dat=0x7fffffffd040, hdl_dat=0x7fffffffcf80, dwg=0x7fffffffd1c0)
[#8] 0x555555632e2d → decode_R2007(dat=0x7fffffffd040, dwg=0x7fffffffd1c0)
[#9] 0x5555555bcebd → dwg_decode(dat=0x7fffffffd040, dwg=0x7fffffffd1c0)
──────
gef➤  p i
$1 = 0x1a

```

reproduce 

./configure --enable-release
make 
./dwgread [poc]

[stackoverflow_dwg_section_wtype_1.zip](https://github.com/LibreDWG/libredwg/files/6603086/stackoverflow_dwg_section_wtype_1.zip)

rurban, 2021-06-06:
Only with illegal input, yes. No proper section name is > 24

rurban, 2021-06-07:
Fixed with ecf5183 (master)

closed 2021-06-07 by rurban