LibreDWG/libredwg #356: Heap use after free in dwg_encode_add_object in src/encode.c:3877

giantbranch, 2021-06-09:
## Affected version 

the latest commit: https://github.com/LibreDWG/libredwg/tree/45d2a290c65ed691be0901ba2b2ef51044e07a16 and 0.12.4

## What's the problem?

A heap use after free was discovered in dwg_encode_add_object in src/encode.c:3877.

ASAN report:

```
$ ./dwgrewrite ./pocs/tests_64231
==25674==ERROR: AddressSanitizer: heap-use-after-free on address 0x7f9c6f40599f at pc 0x00000049511f bp 0x7ffc234960f0 sp 0x7ffc234958b8
READ of size 178195 at 0x7f9c6f40599f thread T0
    #0 0x49511e in __asan_memmove (/root/fuzz/dwgrewrite+0x49511e)
    #1 0x1016996 in memmove /usr/include/x86_64-linux-gnu/bits/string3.h:59:10
    #2 0x1016996 in dwg_encode_add_object /root/fuzz/45d2a29/src/encode.c:3877:11
    #3 0xfc8b60 in dwg_encode /root/fuzz/45d2a29/src/encode.c:2352:16
    #4 0x4c8a03 in dwg_write_file /root/fuzz/45d2a29/src/dwg.c:429:11
    #5 0x4c605a in main /root/fuzz/45d2a29/programs/dwgrewrite.c:349:11
    #6 0x7f9c7f3b283f in __libc_start_main /build/glibc-S7Ft5T/glibc-2.23/csu/../csu/libc-start.c:291
    #7 0x41ba18 in _start (/root/fuzz/dwgrewrite+0x41ba18)

0x7f9c6f40599f is located 172447 bytes inside of 138439591-byte region [0x7f9c6f3db800,0x7f9c777e23a7)
freed by thread T0 here:
    #0 0x4955ed in free (/root/fuzz/dwgrewrite+0x4955ed)
    #1 0x4e36c4 in bit_chain_free /root/fuzz/45d2a29/src/bits.c:3043:7
    #2 0x130fd7e in dwg_encode_PLANESURFACE /root/fuzz/45d2a29/src/./dwg.spec:8818:1
    #3 0x1035f5e in dwg_encode_variable_type /root/fuzz/45d2a29/src/./classes.inc:243:3

previously allocated by thread T0 here:
    #0 0x495b89 in realloc (/root/fuzz/dwgrewrite+0x495b89)
    #1 0x4d508b in bit_chain_alloc /root/fuzz/45d2a29/src/bits.c:3026:30
    #2 0x4c8a03 in dwg_write_file /root/fuzz/45d2a29/src/dwg.c:429:11
    #3 0x4c605a in main /root/fuzz/45d2a29/programs/dwgrewrite.c:349:11
    #4 0x7f9c7f3b283f in __libc_start_main /build/glibc-S7Ft5T/glibc-2.23/csu/../csu/libc-start.c:291

SUMMARY: AddressSanitizer: heap-use-after-free (/root/fuzz/dwgrewrite+0x49511e) in __asan_memmove
Shadow bytes around the buggy address:
  0x0ff40de78ae0: fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd
  0x0ff40de78af0: fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd
  0x0ff40de78b00: fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd
  0x0ff40de78b10: fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd
  0x0ff40de78b20: fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd
=>0x0ff40de78b30: fd fd fd[fd]fd fd fd fd fd fd fd fd fd fd fd fd
  0x0ff40de78b40: fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd
  0x0ff40de78b50: fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd
  0x0ff40de78b60: fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd
  0x0ff40de78b70: fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd
  0x0ff40de78b80: fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
  Shadow gap:              cc
==25674==ABORTING

```

## Compile command

```
CC="clang" CFLAGS="-O1 -g -fsanitize=address" ./configure --enable-release --disable-shared && make
```

## How can we reproduce the issue?

```
unzip tests_64231.zip
dwgbmp ./tests_64231
```
POC file :  
[tests_64231.zip](https://github.com/LibreDWG/libredwg/files/6621939/tests_64231.zip)

rurban, 2022-02-06:
Same problem as #364, #361 and #360

rurban, 2023-06-27:
Fixed with 69dcecd2157b467ddffb94c0d01b6d6e0c1b3b62

closed 2023-06-27 by rurban