LibreDWG/libredwg #357: Null pointer deference in function remove_NOD_item in encode.c:1361

giantbranch, 2021-06-09:
## Affected version 

the latest commit: https://github.com/LibreDWG/libredwg/tree/45d2a290c65ed691be0901ba2b2ef51044e07a16 and 0.12.4

## What's the problem?

A null pointer deference was discovered in remove_NOD_item  in src/encode.c:1361.

ASAN report:

```
$ ./dwgrewrite ./pocs/tests_64231
......
......
......
=================================================================
==25781==ERROR: AddressSanitizer: SEGV on unknown address 0x000000000018 (pc 0x000001041b8f bp 0x60e000000120 sp 0x7ffe1389dbb0 T0)
==25781==The signal is caused by a READ memory access.
==25781==Hint: address points to the zero page.
    #0 0x1041b8f in remove_NOD_item /root/fuzz/45d2a29/src/encode.c:1361:12
    #1 0xfe46d5 in fixup_NOD /root/fuzz/45d2a29/src/encode.c
    #2 0xfc20a3 in dwg_encode /root/fuzz/45d2a29/src/encode.c:1849:21
    #3 0x4c8a03 in dwg_write_file /root/fuzz/45d2a29/src/dwg.c:429:11
    #4 0x4c605a in main /root/fuzz/45d2a29/programs/dwgrewrite.c:349:11
    #5 0x7f3f93b5683f in __libc_start_main /build/glibc-S7Ft5T/glibc-2.23/csu/../csu/libc-start.c:291
    #6 0x41ba18 in _start (/root/fuzz/dwgrewrite+0x41ba18)

AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV /root/fuzz/45d2a29/src/encode.c:1361:12 in remove_NOD_item
==25781==ABORTING

```

## Compile command

```
CC="clang" CFLAGS="-O1 -g -fsanitize=address" ./configure --enable-release --disable-shared && make
```

## How can we reproduce the issue?

```
unzip tests_64230.zip
dwgbmp ./tests_64230
```
POC file :  

[tests_64230.zip](https://github.com/LibreDWG/libredwg/files/6621998/tests_64230.zip)

rurban, 2022-02-06:
After fixing the NOD problem, same problem as GH #364, #361, #360 and #358

rurban, 2022-12-15:
Handled in #497

closed 2022-12-15 by rurban