LibreDWG/libredwg #367: Null pointer deference in function dxf_read_CMC in in_dxf.c

giantbranch, 2021-06-10:
## Affected version 

the latest commit: https://github.com/LibreDWG/libredwg/tree/45d2a290c65ed691be0901ba2b2ef51044e07a16 and 0.12.4

## What's the problem?

A null pointer deference was discovered in function dxf_read_CMC in in_dxf.c.

ASAN report:

```
==2463508==ERROR: AddressSanitizer: SEGV on unknown address 0x000000000000 (pc 0x000001b314cb bp 0x7fffffffbad0 sp 0x7fffffffba80 T0)
==2463508==The signal is caused by a READ memory access.
==2463508==Hint: address points to the zero page.
    #0 0x1b314cb in dxf_read_CMC /src/libredwg/src/in_dxf.c:980:13
    #1 0x1b319f9 in dxf_read_CMC /src/libredwg/src/in_dxf.c:999:15
    #2 0x1b10909 in add_AcDbDetailViewStyle /src/libredwg/src/in_dxf.c:8124:3
    #3 0x1a946f0 in new_object /src/libredwg/src/in_dxf.c:10600:22
    #4 0x1a78816 in dxf_objects_read /src/libredwg/src/in_dxf.c:12301:22
    #5 0x1a6d0f9 in dwg_read_dxf /src/libredwg/src/in_dxf.c:12838:23
    #6 0x4cb82a in dxf_read_file /src/libredwg/src/dwg.c:381:13
    #7 0x4c91c4 in main /src/libredwg/programs/dxf2dwg.c:312:15
    #8 0x7ffff7c580b2 in __libc_start_main /build/glibc-eX1tMB/glibc-2.31/csu/../csu/libc-start.c:308:16
    #9 0x41c40d in _start (/mnt/disk/out/libredwg/programs_asan/dxf2dwg+0x41c40d)

AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV /src/libredwg/src/in_dxf.c:980:13 in dxf_read_CMC
==2463508==ABORTING
```

## Compile command

```
CC="clang" CFLAGS="-O1 -g -fsanitize=address" ./configure --enable-release --disable-shared && make
```

## How can we reproduce the issue?

```
unzip tests_64235.zip
dxf2dwg ./tests_64235 -o /dev/null
```
POC file :  
[tests_64235.zip](https://github.com/LibreDWG/libredwg/files/6629712/tests_64235.zip)

giantbranch, 2021-06-10:
attention：

giantbranch, 2021-06-10:
Please note：add_AcDbSectionViewStyle also call dxf_read_CMC，and SEGV on unknown address 0x000000000000

rurban, 2022-02-05:
Fixed in 0.12.5

closed 2022-02-05 by rurban