LibreDWG/libredwg #369: Null pointer deference in function dwg_handle_name in dwg.c

giantbranch, 2021-06-10:
## Affected version 

the latest commit: https://github.com/LibreDWG/libredwg/tree/45d2a290c65ed691be0901ba2b2ef51044e07a16 and 0.12.4

## What's the problem?

A null pointer deference was discovered in function dwg_handle_name in dwg.c.

ASAN report:

```
==15787==ERROR: AddressSanitizer: SEGV on unknown address 0x000000000000 (pc 0x7ffff6f4e7c6 bp 0x7fffffffc950 sp 0x7fffffffc108 T0)
==15787==The signal is caused by a READ memory access.
==15787==Hint: address points to the zero page.
    #0 0x7ffff6f4e7c6 in strlen /build/glibc-S7Ft5T/glibc-2.23/string/../sysdeps/x86_64/strlen.S:106
    #1 0x481fe2 in strdup /src/llvm-project-12.0.0.src/compiler-rt/lib/asan/asan_interceptors.cpp:433:17
    #2 0x4d50ed in dwg_handle_name /src/libredwg/src/dwg.c:2545:30
    #3 0x5c216e in dwg_dim_blockname /src/libredwg/src/decode.c:5998:22
    #4 0xee1c09 in dwg_dxf_DIMENSION_DIAMETER_private /src/libredwg/src/./dwg.spec:1416:3
    #5 0xe9f979 in dwg_dxf_DIMENSION_DIAMETER /src/libredwg/src/./dwg.spec:1414:1
    #6 0xe9655f in dwg_dxf_object /src/libredwg/src/out_dxf.c:2670:14
    #7 0xe6817c in dxf_entities_write /src/libredwg/src/out_dxf.c:3532:16
    #8 0xe41b4c in dwg_write_dxf /src/libredwg/src/out_dxf.c:3682:7
    #9 0x4c864b in main /src/libredwg/programs/dxfwrite.c:451:11
    #10 0x7ffff6ee383f in __libc_start_main /build/glibc-S7Ft5T/glibc-2.23/csu/../csu/libc-start.c:291
    #11 0x41bc98 in _start (/mnt/disk/out/libredwg/programs_asan/dxfwrite+0x41bc98)

AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV /build/glibc-S7Ft5T/glibc-2.23/string/../sysdeps/x86_64/strlen.S:106 in strlen
==15787==ABORTING
```

## Compile command

```
CC="clang" CFLAGS="-O1 -g -fsanitize=address" ./configure --enable-release --disable-shared && make
```

## How can we reproduce the issue?

```
unzip tests_64245.zip
dxfwrite -I DWG -o /dev/null -y ./tests_64245
```
POC file :  
[tests_64245.zip](https://github.com/LibreDWG/libredwg/files/6629947/tests_64245.zip)

franky9911, 2021-07-11:
I downloaded the latest release 0.12.4-win64, use in VS2017, shows <dwg.h> at line 9023, error C2219: syntax error: type qualifier must be after '*', is there missing a * before restrict? It reads dwg file well if I comment that function which happened to be dwg_handle_name(), POSIX complier seems not checking this syntax, I built from source in cygwin64 without '*' and added '*', both succeeded.

rurban, 2022-02-05:
Fixed in 0.12.5

closed 2022-02-05 by rurban