LibreDWG/libredwg #370: Heap-buffer-overflow in add_ASSOCACTION in in_dxf.c

giantbranch, 2021-06-10:
## Affected version 

the latest commit: https://github.com/LibreDWG/libredwg/tree/45d2a290c65ed691be0901ba2b2ef51044e07a16 and 0.12.4

## What's the problem?

A heap buffer overflow was discovered in add_ASSOCACTION  in in_dxf.c.

ASAN report:

```
==24181==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x6040000042b8 at pc 0x00000049718a bp 0x7fffffffbaf0 sp 0x7fffffffb2b8
WRITE of size 48 at 0x6040000042b8 thread T0
    #0 0x497189 in __asan_memcpy /src/llvm-project-12.0.0.src/compiler-rt/lib/asan/asan_interceptors_memintrinsics.cpp:22:3
    #1 0x1ad7ba3 in add_ASSOCACTION /src/libredwg/src/in_dxf.c:5924:23
    #2 0x1a8b7a4 in new_object /src/libredwg/src/in_dxf.c:9331:26
    #3 0x1a78816 in dxf_objects_read /src/libredwg/src/in_dxf.c:12301:22
    #4 0x1a6d0f9 in dwg_read_dxf /src/libredwg/src/in_dxf.c:12838:23
    #5 0x4cb82a in dxf_read_file /src/libredwg/src/dwg.c:381:13
    #6 0x4c91c4 in main /src/libredwg/programs/dxf2dwg.c:312:15
    #7 0x7ffff7c580b2 in __libc_start_main /build/glibc-eX1tMB/glibc-2.31/csu/../csu/libc-start.c:308:16
    #8 0x41c40d in _start (/mnt/disk/out/libredwg/programs_asan/dxf2dwg+0x41c40d)

0x6040000042b8 is located 0 bytes to the right of 40-byte region [0x604000004290,0x6040000042b8)
allocated by thread T0 here:
    #0 0x497eb2 in calloc /src/llvm-project-12.0.0.src/compiler-rt/lib/asan/asan_malloc_linux.cpp:154:3
    #1 0x1a687d9 in xcalloc /src/libredwg/src/in_dxf.c:216:7
    #2 0x1ad7b18 in add_ASSOCACTION /src/libredwg/src/in_dxf.c:5915:38
    #3 0x1a8b7a4 in new_object /src/libredwg/src/in_dxf.c:9331:26
    #4 0x1a78816 in dxf_objects_read /src/libredwg/src/in_dxf.c:12301:22
    #5 0x1a6d0f9 in dwg_read_dxf /src/libredwg/src/in_dxf.c:12838:23
    #6 0x4cb82a in dxf_read_file /src/libredwg/src/dwg.c:381:13
    #7 0x4c91c4 in main /src/libredwg/programs/dxf2dwg.c:312:15
    #8 0x7ffff7c580b2 in __libc_start_main /build/glibc-eX1tMB/glibc-2.31/csu/../csu/libc-start.c:308:16

SUMMARY: AddressSanitizer: heap-buffer-overflow /src/llvm-project-12.0.0.src/compiler-rt/lib/asan/asan_interceptors_memintrinsics.cpp:22:3 in __asan_memcpy
Shadow bytes around the buggy address:
  0x0c087fff8800: fa fa 00 00 00 00 00 fa fa fa 00 00 00 00 00 fa
  0x0c087fff8810: fa fa 00 00 00 00 00 fa fa fa 00 00 00 00 00 fa
  0x0c087fff8820: fa fa 00 00 00 00 00 fa fa fa 00 00 00 00 00 fa
  0x0c087fff8830: fa fa 00 00 00 00 00 fa fa fa 00 00 00 00 00 fa
  0x0c087fff8840: fa fa 00 00 00 00 00 fa fa fa 00 00 00 00 00 fa
=>0x0c087fff8850: fa fa 00 00 00 00 00[fa]fa fa 00 00 00 00 00 00
  0x0c087fff8860: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c087fff8870: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c087fff8880: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c087fff8890: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c087fff88a0: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07 
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
  Shadow gap:              cc
==24181==ABORTING
```

## Compile command

```
CC="clang" CFLAGS="-O1 -g -fsanitize=address" ./configure --enable-release --disable-shared && make
```

## How can we reproduce the issue?

```
unzip tests_64248.zip
dxf2dwg ./tests_64248 -o /dev/null
```
POC file :  

[tests_64248.zip](https://github.com/LibreDWG/libredwg/files/6630146/tests_64248.zip)

rurban, 2021-06-27:
This is a good catch, unrelated to fuzzed input. The VALUEPARAM allocation for values[] was wrong

rurban, 2022-02-05:
Fixed in 0.12.5

closed 2022-02-05 by rurban