LibreDWG/libredwg #394: dwg2dxf conversion is not converting all informations.

vchoudharyb, 2021-09-08:
Hi, I have attached three files. one original DWG file and two dxf files one converted using libreDWG and other using online converter. The online converted file is having all information where as where as dxf file converted using libredwg is not having all information, some information are missing. Can you please fix it?
[file to share.zip](https://github.com/LibreDWG/libredwg/files/7127936/file.to.share.zip)

rurban, 2021-09-09:
Missing:
* $TDUCREATE, $TDUUPDATE
* AcDbPlaceHolder, AcDbLayout classes
* SPLINE 40
* ACAD_PROXY_OBJECT

Different:
* $MEASUREMENT (256 => 1)
* SPLINE 70, 42,43 ordering

We don't support proxies yet, we rather save the original object. cloudconvert did not understand many objects, like ViewStyles.

vchoudharyb, 2021-09-14:
When should I expect this enhancement implemented as a part of this library?

rurban, 2022-02-04:
No idea. It gets faster if someone else helps

michal-josef-spacek, 2023-03-26:
Files in this issue:
* Converted using libredwg library.dxf - AC1018
* Converted using other tool.dxf - AC1018
* Original file.DWG - AC1021 (r2007)

rurban, 2023-03-26:
What is missing?

michal-josef-spacek, 2023-03-26:
> What is missing?

There is a checkbox with missing items from you :-) 

I was looking at it and there are many differences. I am working on older versions still