LibreDWG/libredwg #396: ./dwg.spec:1557:17: warning: format specifies type 'unsigned short' but the argument has type 'int' [-Wformat]

no-such-user, 2021-11-23:
I'm seeing a couple of the above messages when compiling on a Mac. Does it matter?

./dwg.spec:7449:9: warning: format specifies type 'unsigned short' but the argument has type 'BITCODE_BL' (aka 'unsigned int') [-Wformat]

no-such-user, 2021-11-23:
interestingly, all tests passed:
```
PASS: check-objects.pl
====================
All 247 tests passed
====================

```

but MAKEINFO failed on "make check", but not on "make". It had failed the same way on make, but I installed a newer textinfo

`brew install texinfo
`
Seems to be working, though.

rurban, 2021-11-24:
What kind of compiler target exactly, 32bit or 64bit?

no-such-user, 2021-11-24:
% ./configure
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a race-free mkdir -p... ./build-aux/install-sh -c -d
checking for gawk... no
checking for mawk... no
checking for nawk... no
checking for awk... awk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether make supports nested variables... (cached) yes
checking git version... 0.12.4.4367_dirty
checking so version-info... 0:12:0
checking build system type... x86_64-apple-darwin20.6.0
checking host system type... x86_64-apple-darwin20.6.0
./configure: line 3888: PKG_PROG_PKG_CONFIG: command not found
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether the compiler supports GNU C... yes
checking whether gcc accepts -g... yes
checking for gcc option to enable C11 features... none needed
checking whether gcc understands -c and -o together... yes
checking whether make supports the include directive... yes (GNU style)
checking dependency style of gcc... gcc3
checking whether ln -s works... yes
checking whether make sets $(MAKE)... (cached) yes
checking how to print strings... printf
checking for a sed that does not truncate output... /usr/bin/sed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for fgrep... /usr/bin/grep -F
checking for ld used by gcc... /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ld
checking if the linker (/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ld) is GNU ld... no
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface... BSD nm
checking the maximum length of command line arguments... 786432
checking how to convert x86_64-apple-darwin20.6.0 file names to x86_64-apple-darwin20.6.0 format... func_convert_file_noop
checking how to convert x86_64-apple-darwin20.6.0 file names to toolchain format... func_convert_file_noop
checking for /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ld option to reload object files... -r
checking for objdump... objdump
checking how to recognize dependent libraries... pass_all
checking for dlltool... no
checking how to associate runtime and link libraries... printf %s\n
checking for ar... ar
checking for archiver @FILE support... no
checking for strip... strip
checking for ranlib... ranlib
checking command to parse /usr/bin/nm -B output from gcc object... ok
checking for sysroot... no
checking for a working dd... /bin/dd
checking how to truncate binary pipes... /bin/dd bs=4096 count=1
checking for mt... no
checking if : is a manifest tool... no
checking for dsymutil... dsymutil
checking for nmedit... nmedit
checking for lipo... lipo
checking for otool... otool
checking for otool64... no
checking for -single_module linker flag... yes
checking for -exported_symbols_list linker flag... yes
checking for -force_load linker flag... yes
checking for stdio.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
[... 202 lines cut ...]
config.status: creating bindings/Makefile
config.status: creating bindings/python/Makefile
config.status: creating test/Makefile
config.status: creating test/xmlsuite/Makefile
config.status: creating test/unit-testing/Makefile
config.status: creating libredwg.pc
config.status: creating bindings/perl/Makefile.PL
config.status: creating programs/dwgfilter
config.status: creating programs/alive.test
config.status: creating programs/dxf.test
config.status: creating programs/json.test
config.status: creating programs/svg.test
config.status: creating logs-all.sh
config.status: creating log
config.status: creating dwg
config.status: creating dxf
config.status: creating json
config.status: creating svg
config.status: creating rw
config.status: creating dynapi_test_all.sh
config.status: creating unit_testing_all.sh
config.status: creating src/config.h
config.status: src/config.h is unchanged
config.status: executing depfiles commands
config.status: executing libtool commands

rurban, 2022-02-04:
There was a cast missing. thanks

closed 2022-02-04 by rurban