LibreDWG/libredwg #398: Running precompiled version 0.12.4.4367

Nikhilsp, 2022-01-21:
I am not a programmer. Just trying to run an executable file on windows 10. Can I anyone please tell me how to run the exe files in libredwg-0.12.4.4367-win64.zip. Do I need to install some other software to run the exe or download the source code too?

Nikhilsp, 2022-01-21:
Any sugesstions? I tried running the exe files and I am not seeing any dialog box or anything. Just a cmd screen pops and closes.

rurban, 2022-01-21:
You are probably missing the C runtime DLL's. Those are usually called MSVC Restributable.

It was compiled with the runtime for Microsoft Windows Server 2019, 10.0-17763 on Github actions.
Since I dont have a windows PC you have to find these DLL's by yourself. We are also not allowed to restribute it.

Nikhilsp, 2022-01-21:
Thank you, I have installed all the versions of re-distributable from Microsoft. Still getting the same output.

Nikhilsp, 2022-01-21:
works now

closed 2022-01-22 by rurban