LibreDWG/libredwg #404: 0.12.5: No rule to make target 'dwg2ps.1', needed by 'all-am'.  Stop.

AndrewAmmerlaan, 2022-02-10:
Version 0.12.5 release tarball fails to compile for me on Gentoo Linux:

```
make[2]: *** No rule to make target 'dwg2ps.1', needed by 'all-am'.  Stop.
make[2]: *** Waiting for unfinished jobs....
make[2]: Leaving directory '/var/tmp/portage/media-gfx/libredwg-0.12.5/work/libredwg-0.12.5/programs'
make[1]: *** [Makefile:674: all-recursive] Error 1
make[1]: Leaving directory '/var/tmp/portage/media-gfx/libredwg-0.12.5/work/libredwg-0.12.5'
make: *** [Makefile:557: all] Error 2
```

Full build log: [build.log](https://github.com/LibreDWG/libredwg/files/8040242/build.log)

On the other hand, cloning the git repo and compiling that works.

sudhan-c, 2022-02-10:
Hey do u know to add this package in python..

rurban, 2022-02-11:
libps is missing, causing dwg2ps not to be built, causing this wrong manpage error.

AndrewAmmerlaan, 2022-02-11:
> libps is missing, causing dwg2ps not to be built, causing this wrong manpage error.

`pslib-0.4.5` is installed, is this the wrong version? What confuses me is that the release tarball fails but the build from a `git clone` works.

rurban, 2022-02-12:
Any libps version is good. I'll check. 
make distcheck works, which should catch such errors.

AndrewAmmerlaan, 2022-02-15:
> Any libps version is good. I'll check. make distcheck works, which should catch such errors.

`make distcheck` doesn't work for me. On the release tarball I get the same error:
```
make  dist-xz dist-gzip am__post_remove_distdir='@:'
make[1]: Entering directory '/tmp/libredwg-0.12.5'
make  distdir-am
make[2]: Entering directory '/tmp/libredwg-0.12.5'
if test -d "libredwg-0.12.5"; then find "libredwg-0.12.5" -type d ! -perm -200 -exec chmod u+w {} ';' && rm -rf "libredwg-0.12.5" || { sleep 5 && rm -rf "libredwg-0.12.5"; }; else :; fi
test -d "libredwg-0.12.5" || mkdir "libredwg-0.12.5"
(cd src && make  top_distdir=../libredwg-0.12.5 distdir=../libredwg-0.12.5/src \
am__remove_distdir=: am__skip_length_check=: am__skip_mode_fix=: distdir)
make[3]: Entering directory '/tmp/libredwg-0.12.5/src'
make  distdir-am
make[4]: Entering directory '/tmp/libredwg-0.12.5/src'
make[4]: Leaving directory '/tmp/libredwg-0.12.5/src'
make[3]: Leaving directory '/tmp/libredwg-0.12.5/src'
(cd programs && make  top_distdir=../libredwg-0.12.5 distdir=../libredwg-0.12.5/programs \
am__remove_distdir=: am__skip_length_check=: am__skip_mode_fix=: distdir)
make[3]: Entering directory '/tmp/libredwg-0.12.5/programs'
make  distdir-am
make[4]: Entering directory '/tmp/libredwg-0.12.5/programs'
make[4]: *** No rule to make target 'dwg2ps.1', needed by 'distdir-am'.  Stop.
make[4]: Leaving directory '/tmp/libredwg-0.12.5/programs'
make[3]: *** [Makefile:1027: distdir] Error 2
make[3]: Leaving directory '/tmp/libredwg-0.12.5/programs'
make[2]: *** [Makefile:878: distdir-am] Error 1
make[2]: Leaving directory '/tmp/libredwg-0.12.5'
make[1]: *** [Makefile:872: distdir] Error 2
make[1]: Leaving directory '/tmp/libredwg-0.12.5'
make: *** [Makefile:976: dist] Error 2
```

It looks like `dwg2ps.1` is simply absent from the release tar.xz and tar.gz! It is present in the git sources though, which is why that does work. Perhaps there is something wrong with the release tarball generation?

On the git sources I get a different error (but this is a different issue that should be easy to fix):
```
make  dist-xz dist-gzip am__post_remove_distdir='@:'
make[1]: Entering directory '/tmp/libredwg'
make  distdir-am
make[2]: Entering directory '/tmp/libredwg'
if test -d "libredwg-0.12.4.4404_dirty"; then find "libredwg-0.12.4.4404_dirty" -type d ! -perm -200 -exec chmod u+w {} ';' && rm -rf "libredwg-0.12.4.4404_dirty" || { sleep 5 && rm -rf "libredwg-0.12.4.4404_dirty"; }; else :; fi
test -d "libredwg-0.12.4.4404_dirty" || mkdir "libredwg-0.12.4.4404_dirty"
(cd src && make  top_distdir=../libredwg-0.12.4.4404_dirty distdir=../libredwg-0.12.4.4404_dirty/src \
am__remove_distdir=: am__skip_length_check=: am__skip_mode_fix=: distdir)
make[3]: Entering directory '/tmp/libredwg/src'
make  distdir-am
make[4]: Entering directory '/tmp/libredwg/src'
echo make regen-dynapi might be needed
make regen-dynapi might be needed
make[4]: Leaving directory '/tmp/libredwg/src'
make[3]: Leaving directory '/tmp/libredwg/src'
(cd programs && make  top_distdir=../libredwg-0.12.4.4404_dirty distdir=../libredwg-0.12.4.4404_dirty/programs \
am__remove_distdir=: am__skip_length_check=: am__skip_mode_fix=: distdir)
make[3]: Entering directory '/tmp/libredwg/programs'
make  distdir-am
make[4]: Entering directory '/tmp/libredwg/programs'
make[4]: Leaving directory '/tmp/libredwg/programs'
make[3]: Leaving directory '/tmp/libredwg/programs'
(cd examples && make  top_distdir=../libredwg-0.12.4.4404_dirty distdir=../libredwg-0.12.4.4404_dirty/examples \
am__remove_distdir=: am__skip_length_check=: am__skip_mode_fix=: distdir)
make[3]: Entering directory '/tmp/libredwg/examples'
make  distdir-am
make[4]: Entering directory '/tmp/libredwg/examples'
make[4]: Leaving directory '/tmp/libredwg/examples'
make[3]: Leaving directory '/tmp/libredwg/examples'
(cd test && make  top_distdir=../libredwg-0.12.4.4404_dirty distdir=../libredwg-0.12.4.4404_dirty/test \
am__remove_distdir=: am__skip_length_check=: am__skip_mode_fix=: distdir)
make[3]: Entering directory '/tmp/libredwg/test'
make  distdir-am
[... 19 lines cut ...]
make[3]: Entering directory '/tmp/libredwg/bindings'
make  distdir-am
make[4]: Entering directory '/tmp/libredwg/bindings'
(cd python && make  top_distdir=../../libredwg-0.12.4.4404_dirty distdir=../../libredwg-0.12.4.4404_dirty/bindings/python \
am__remove_distdir=: am__skip_length_check=: am__skip_mode_fix=: distdir)
make[5]: Entering directory '/tmp/libredwg/bindings/python'
make  distdir-am
make[6]: Entering directory '/tmp/libredwg/bindings/python'
make[6]: Leaving directory '/tmp/libredwg/bindings/python'
make[5]: Leaving directory '/tmp/libredwg/bindings/python'
make[4]: Leaving directory '/tmp/libredwg/bindings'
make[3]: Leaving directory '/tmp/libredwg/bindings'
(cd doc && make  top_distdir=../libredwg-0.12.4.4404_dirty distdir=../libredwg-0.12.4.4404_dirty/doc \
am__remove_distdir=: am__skip_length_check=: am__skip_mode_fix=: distdir)
make[3]: Entering directory '/tmp/libredwg/doc'
make  distdir-am
Makefile:415: *** Recursive variable 'TEXINPUTS' references itself (eventually).  Stop.
make[3]: Leaving directory '/tmp/libredwg/doc'
make[2]: *** [Makefile:878: distdir-am] Error 1
make[2]: Leaving directory '/tmp/libredwg'
make[1]: *** [Makefile:872: distdir] Error 2
make[1]: Leaving directory '/tmp/libredwg'
make: *** [Makefile:980: dist] Error 2
```
Removing the reference to itself in `TEXINPUTS` and using += instead, fixes the issue.

rurban, 2022-04-23:
should be fixed now

closed 2022-04-23 by rurban