LibreDWG/libredwg #412: unknown_s[] fields in header

michal-josef-spacek, 2022-02-17:
in src/header.spec is section:
```
    for (i = 0; i < 3; i++) { // 3, 5, 205
      FIELD_RS (unknown_s[i], 0);
    }
```

I believe that this three values are numbers of ?, sections and header variables (idea, but i counted and seems that is).
On this values depends header variables (that's true).

3: ?
5: Number of sections (BLOCK, LAYER, STYLE, LTYPE, VIEW)
>=74 AND <=205: Number of header variables (from insertion_base, i don't know if 3d point is one or three)

In AC1.50 DWG files: 3, 5, 74; 3, 5, 83; 
In AC2.10 example (in [BLANK.DWG.gz](https://github.com/LibreDWG/libredwg/files/8088649/BLANK.DWG.gz)): 3, 5, 83
In AC1001 DWG files: 3, 5, 101
In AC1002 DWG files: 3, 5, 104; 3, 5, 114
In AC1003 DWG files: 3, 5, 120; 3, 5, 122
In AC1004 DWG files: 3, 5, 129
In AC1006 DWG files: 3, 5, 158; 3, 5, 160
In AC1009 DWG files: 3, 5, 204; 3, 5, 205

michal-josef-spacek, 2022-02-17:
I found this idea with count of some now. When i am writing this issue :-D So i need to precise. Probably right (not count of bytes or something like this). I checked between 3,5,74 and 3,5,83 (header place in 3,5,83 is plus 0x01fa-0x02e4)

rurban, 2022-02-17:
yes, looks convincing

michal-josef-spacek, 2022-02-17:
@rurban Thanks for code, i will check it.

michal-josef-spacek, 2022-02-17:
@rurban I checked AC1.50 and there is 74 variables in header, so idea is right. 3d point is three variables.
FYI: Last variable for AC1.50 is $DIMTAD.

michal-josef-spacek, 2022-02-18:
@rurban We maybe need to count variables in header and stop parsing if count is greater than num_header_vars. 
We need to detect variables in header still. I have AC1.50, i will fill issues for it.

rurban, 2022-02-18:
Yes, something like this. Still searching for my old parser ...

michal-josef-spacek, 2022-04-01:
We could close this ticket. Third number is number of header variables definitively.

rurban, 2022-04-14:
BTW: Found my old dwg11.c parser: https://gist.github.com/rurban/8c3a65ad8f75ff858c900392b7773dd0#file-dwg11-c

closed 2022-04-14 by rurban

michal-josef-spacek, 2022-04-14:
@rurban Great, thank you.