LibreDWG/libredwg #422: CMake build issue: MinGW does not support memmem()

carmethene, 2022-03-02:
MinGW links the MSVC CRT rather than glibc, thus memmem() is not available.  It appears that an alternative is already provided in src/common.h, but this is never used when building through CMake as cmakeconfig.h.in #defines HAVE_MEMMEM as 1 in all cases.

CMake has a utility which easily resolves this issue:

In CMakeLists.txt:
```
include(CheckSymbolExists)
check_symbol_exists("memmem" "string.h" HAVE_MEMMEM)
```

In cmakeconfig.h.in:
```
/* Define if you have the `memmem' function. */
#cmakedefine HAVE_MEMMEM
```

carmethene, 2022-03-02:
Just realised I'm adding to your workload, so I'll make a pull request :)

rurban, 2022-03-03:
Nice, but beware the legal issues. Stay under 15 lines, or send a copyright assignment. Seems to be very fast now.

commit 93ecf517 (2022-03-04): Fix #422 - Missing functions in CMake/MinGW

- Correctly set HAVE_* #defines in configuration step so that fallback options are used when these functions are not available.
closed 2022-03-04 by rurban