LibreDWG/libredwg #424: Disable logging a lot of info in console

GeorgGrebenyuk, 2022-03-06:
Hi, i have a little question - how disable logging info in console? I have reading [that post](https://github.com/LibreDWG/libredwg/issues/25#issuecomment-399935076), but couldn't understood what need did. I work in MS VS's environmental*
[image]

carmethene, 2022-03-09:
Dwg_Data has a member which controls the log level.  If you don't initialize that struct, it'll get some random value which will likely enable all the debug logging.  Simple fix:

`memset(&my_dwg, 0, sizeof(Dwg_Data));
`

GeorgGrebenyuk, 2022-03-10:
> memset(&my_dwg, 0, sizeof(Dwg_Data));

Yeah, thank you -- it's variable that I wanted find! (it was placed more deeper that I thought 😅)

closed 2022-03-10 by GeorgGrebenyuk