LibreDWG/libredwg #425: Potential security issue

JamieSlome, 2022-03-23:
Hey there!

I belong to an open source security research community, and a member (@peacock-doris) has found an issue, but doesn’t know the best way to disclose it.

If not a hassle, might you kindly add a `SECURITY.md` file with an email, or another contact method? GitHub [recommends](https://docs.github.com/en/code-security/getting-started/adding-a-security-policy-to-your-repository) this best practice to ensure security issues are responsibly disclosed, and it would serve as a simple instruction for security researchers in the future.

Thank you for your consideration, and I look forward to hearing from you!

(cc @huntr-helper)

rurban, 2022-03-23:
send it to me, @rurban as mentioned in SECURITY.md

Reini

JamieSlome, 2022-03-24:
@rurban - sent to your e-mail on your GitHub profile 👍 

Just for reference, the report can be found here:
https://huntr.dev/bounties/4ada4519-c3b6-4814-b822-65dcb33cf98d/

It is private and only accessible to maintainers with repository write permissions, if not using the protected URL sent in the e-mail.

rurban, 2022-03-24:
Hi
Can you ask peackock-doris, with which commit exactly she found it? In my current master the line numbers don't matchup.

JamieSlome, 2022-03-24:
Dropped a comment on the report @rurban - thanks 👍

JamieSlome, 2022-03-25:
@rurban - the researcher has specified the commit on the report 👍

rurban, 2022-04-14:
Fixed in master thanks

closed 2022-04-14 by rurban