LibreDWG/libredwg #427: How to test pre12 DWG files?

michal-josef-spacek, 2022-04-01:
Good way is create script files (*.scr and script command in autocad), which could be used for generation of DWG/DXF files.
I could prepare some DWG files. 

But i don't understand testing system for it. 
@rurban could you help me with it?

rurban, 2022-04-14:
The major question is copyright. Both of use do have many <r13 dwg's, but we may not publish them.
A proper file converter would be good, so we can have example_r9,... dwg's.
Or I finish the preR13 encoder and use these files then for the tests.

michal-josef-spacek, 2022-04-15:
@rurban In my testing suite in kaitai projects is bunch of DWG files generated by me from real AutoCADs. I created DWG file which has 2d point, DWG file which has 3d point, DWG file for each version of header file value. 

There were created by real AutoCADs, this is ok from copyright side or not?

Example (FRONTZ1.scr):
```
line 0,0 5,5 
dview l  cl f 1.2345 x
```
autocad_command: script FRONTZ1.scr generate DWG example (AC1006) for FRONTZ1 variable with 1.2345 value.

This is good, because i have test case for FRONTZ1 with 0 and with 1.2345 and will test it precisly.

michal-josef-spacek, 2022-04-15:
In best way: test e.g. string in maximal length, number in maximal value etc.

rurban, 2022-04-16:
Good. then we just need the DWG's added to test-data. But best add example_rx.dwg's with all available entities and blocks for full coverage.

michal-josef-spacek, 2022-04-16:
@rurban ok, i will look to it and create example dwg file for some version. We will see.

michal-josef-spacek, 2022-04-20:
First DWG file: https://github.com/LibreDWG/libredwg/pull/438

rurban, 2022-04-21:
Crazy DXF format!

michal-josef-spacek, 2022-04-21:
@rurban yes :-D

michal-josef-spacek, 2022-04-23:
Test data for AC2.10: https://github.com/LibreDWG/libredwg/pull/440

michal-josef-spacek, 2022-04-24:
Test data for AC1003: https://github.com/LibreDWG/libredwg/pull/441

rurban, 2022-05-05:
In the smoke/preR13-iterators branch I've added support to test all the new DWG's/DXF with our old tests.
Still some memory leaks, and missing indxf, otherwise it works fine now

closed 2022-05-05 by rurban