LibreDWG/libredwg #429: dwg2SVG

Takieddinex, 2022-04-06:
Hello,
I would like to use the library already compiled in Windows but when I enter the command dwg2SVG it returns empty svg output

Could you help me please? 
[image]
[image]
[image]

rurban, 2022-04-06:
pasting Images does not really help.
you need to leave out the [] part, this means optional. verbosity 0- 9.

closed 2022-04-06 by rurban

rurban, 2022-04-06:
inspect the bet.svg for any errors.
or use -v3

Takieddinex, 2022-04-06:
"This XML file does not appear to have any style information associated with it. The document tree is shown below."

I'm doing the inspection but I only find this message in span 
I tried -v3 but it doesn't work

Takieddinex, 2022-04-19:
Hello ! 
I can't find the method to do the conversion if you can please help me 

./dwg2SVG.exe [-v[0-9]] D:/libredwg/projet.dwg>D:/libredwg/test.svg

this is the command I enter and I would get an empty svg file as output 

[image]

Thank you !!