LibreDWG/libredwg #435: Fix of handling if in entity

michal-josef-spacek, 2022-04-19:
We need to fix handling id in entity.
Stub here: https://github.com/LibreDWG/libredwg/commit/2a92ef3dcd151ef65d09a3e00334abcd453846cd
This is length and string without termination as i understang.

Example:
[POINT1.DXF.gz](https://github.com/LibreDWG/libredwg/files/8516065/POINT1.DXF.gz)
[POINT1.DWG.gz](https://github.com/LibreDWG/libredwg/files/8516066/POINT1.DWG.gz)

rurban, 2022-05-10:
No, it's an ordinary handle as after R13. Just no code for ownership, only size and value.
Fixed in work/preR13

rurban, 2022-05-11:
No, 24 should be 0.1.36 (for 0x24)

michal-josef-spacek, 2022-12-22:
Actuall output is:
```
type: 2 [RCd]
Add entity POINT [14] Decode entity POINT
===========================
Entity number: 14, Type: 2, Addr: 7ea
flag_r11: 0x64 [RC 0]
          ELEVATION (04) HANDLING (20) PSPACE (40)
size: 31 [RS]
layer: 0 [H(RSd) 8]
opts_r11: 0x0 [RSx 0]
viewport: 4 [H(RC) 0]
handle: 0.1.24 [H 5]
extra_r11: 1 [RSd 0]
x: 1.000000 [RD 10]
y: 1.000000 [RD 20]
crc: DBD3 [RSx]
```

rurban, 2022-12-22:
This looks good to me. 24 matches now the dxf

michal-josef-spacek, 2023-01-06:
> No, 24 should be 0.1.36 (for 0x24)

You are saying that the handle is in decimal form. Are you sure?

I am looking to log output and that's in hexadecimal form.
If it's ok, we could close this issue.

closed 2023-01-06 by rurban